WebODM/WebODM #1: User Interface

mojodna, 2016-08-15:
The WebODM user interface needs to surface the following functionality:

* task management - creation and status
* model display - allow orthophotos + 3D models to be browsed
* cluster management - ???
* mission planning - is there existing FOSS for this?
* documentation

@pierotofy can you clarify what you're thinking around cluster management?

I'm also thinking that model display and mission planning might be better suited as their own applications, integrating with the WebODM API according to their needs.

Thoughts?

pierotofy, 2016-08-15:
By "cluster management" (there's probably a better term for this) I mean the management of distributing image processing tasks across multiple server instances.

This image might clarify a bit:

[image]

The Web Manager is WebODM.

The REST API is https://github.com/pierotofy/node-OpenDroneMap, http://nodeodm.masseranolabs.com/

Mission planning could be added as a plugin if we don't want to add too many functionalities. (This would be one of the final items to develop anyway, certainly not a priority).

smathermather, 2016-08-16:
Existing FOSS Mission Planning: http://ardupilot.org/planner/index.html

pierotofy, 2016-08-16:
Yup. Are there libraries that would facilitate path planning (ex. given a polygon, an overlap % and an angle, returns the line string that covers that particular polygon)? 

I found this link time ago that might help: http://www.engr.usask.ca/classes/GEOE/218/notes/airphoto_reading/apg.htm

vesnikos, 2016-09-02:
Pix4d have created some very nice articles/guidelines for UAVs' flightpaths that are going to capture images for sfm-mvs purposes. 

https://support.pix4d.com/hc/en-us/articles/202557459-Step-1-Before-Starting-a-Project-1-Designing-the-Image-Acquisition-Plan-a-Selecting-the-Image-Acquisition-Plan-Type#gsc.tab=0

https://support.pix4d.com/hc/en-us/articles/202557479#gsc.tab=0

> Are there libraries that would facilitate path planning (ex. given a polygon, an overlap % and an angle, returns the line string that covers that particular polygon)?

None that I am aware of, but given the bbox of an Area of Interest and the images footprint is very easy to determine where the images should be taken at for a desired overlap. This can be facilitated directly at the database if its spatially enabled (e.g. postgis)

pierotofy, 2016-09-04:
Great links, thanks @vesnikos!

smathermather, 2016-09-06:
It's true @vesnikos -- with SFCGAL enabled PostGIS, you can do some pretty fancy stuff in the database:
https://smathermather.wordpress.com/2013/12/15/uas-drone-footprint-geometries-calculated-in-postgis-with-sfcgal-for-real-this-time/

Bavar2142, 2016-10-03:
Being able to use distributed processing would really help those of us that dont have the  power in house. Would it be anything like boinc where we could donate spare processing capacity to the network?

pierotofy, 2016-10-03:
@Bavar2142 yes! At some point we should try to create a network of free processing nodes, run by volunteers.

Bavar2142, 2016-10-03:
Thatd be great. I know id love to donate processing time on my home rig. 4.3 ghz i5 6600k with H110i here. How would we go about that? I could look into that boinc client if you like.

pierotofy, 2016-10-03:
I opened a separate issue #24.

ns-1m, 2016-10-05:
* mission planning - is there existing FOSS for this?

Here's a web based mission planner based on Ardupilot.

Skysense Planner Video - Open-source Web Mission Planner for MAVLink Drones
http://diydrones.com/profiles/blogs/skysense-planner-video-open-source-web-mission-planner-for

http://www.skysense.co/planner-launch

Bavar2142, 2016-10-05:
@ns-1m Would this work with a phantom 3?

ns-1m, 2016-10-05:
@ns-1m Would this work with a phantom 3?

@Bavar2142, No. I think DJI drones does not have a "real" mission planner for mapping.

closed 2019-04-03 by pierotofy