WebODM/WebODM #4: 3D Model Display

mojodna, 2016-08-15:
ODM-generated 3D models should be browsable using a Cesium-like interface.

Needs:

* [Cesium inputs; I don't know anything about these]
* ...

pierotofy, 2016-08-15:
It is possible to display both 3D models and Point Clouds using Cesium, some research and development needs to be done to convert the resulting models to 3D tiles, but it should be feasible: https://cesiumjs.org/Chappes/index.html http://cesiumjs.org/2015/08/10/introducing-3d-tiles/

smathermather, 2016-08-16:
3D tiles don't scale well. See http://speck.ly or http://potree.org/ for really well scaled point clouds. I agree Cesium may be the best option for textured meshes. This I know much less about.

pierotofy, 2016-08-16:
potree.org looks pretty good! Nice find.

smathermather, 2016-08-16:
potree has implemented the octree optimizations necessary for rapid and beautiful viewing of point clouds, which I think is also in the pipeline for what you see at speck.ly.

The ecosystem around https://speck.ly (including https://plas.io, etc.) solves the data storage and delivery end of things, plus some of the display issues.

@hobu -- can you point @mojodna and @pierotofy to best resources here?

hobu, 2016-08-16:
> 3D tiles don't scale well

3D tiles scale just fine. It's just they aren't done yet for point clouds.

spatialhast, 2016-11-27:
Using https://threejs.org/ possible display Textured Mesh using direct loading data (example https://threejs.org/examples/webgl_loader_obj_mtl.html). Example textured mesh data viewer using three.js available on https://spatialhast.github.io/tmviewer/ link. In this example used textured mesh from 'odm_data_toledo' and three.js templates from https://github.com/kaisalmen/WWOBJLoader by @kaisalmen. On viewer https://spatialhast.github.io/tmviewer/index_geo.html displayed data 'black' color - there is used Georeferenced textured mesh.

smathermather, 2016-11-28:
Whoa! This is great info @spatialhast!

pierotofy, 2016-12-14:
Currently looking at http://osgjs.org/, which is the library that's being used by sketchfab for their 3d model viewer.

pierotofy, 2017-01-22:
This has been added as part of #78, thanks everyone for your input!

closed 2017-01-22 by pierotofy

smathermather, 2017-01-24:
w00t!