WebODM/WebODM #7: Cluster Management

mojodna, 2016-08-15:
[What does Cluster Management mean in the context of WebODM?]

From #1:

By "cluster management" (there's probably a better term for this) I mean the management of distributing image processing tasks across multiple server instances.

This image might clarify a bit:

https://cloud.githubusercontent.com/assets/1951843/17680196/9bfe878e-6304-11e6-852e-c09f1e02f3c0.png

The Web Manager is WebODM.

The REST API is https://github.com/pierotofy/node-OpenDroneMap, http://nodeodm.masseranolabs.com/

mojodna, 2016-08-15:
@pierotofy the audience for this piece is system administrators?

pierotofy, 2016-08-15:
Yes (only users with admin permissions will get to manage the clusters).

pierotofy, 2016-08-16:
New name for "cluster". `ProcessingNode`. Seems more intuitive.

pierotofy, 2016-10-26:
This can be closed, there's nothing to add here.

closed 2016-10-26 by pierotofy