WebODM/WebODM #9: Ability to overwrite default settings.

vesnikos, 2016-09-11:
An ability to overwrite the default settings is usefull in numerous occasions. 

This can be implemented easily by appending :

> try:
>     from .local_settings import *
> except ImportError:
>     pass
> 

ofcourse local_settings.py  should be added in .gitignore as well

pierotofy, 2016-09-11:
I agree, this should be added. Would you want to take care of this, or should I add it to my TODO list?

pierotofy, 2016-09-11:
Never mind, I added this as part of #10. Thanks!

closed 2016-09-11 by pierotofy

vesnikos, 2016-09-11:
I would suggest to add local_settings.py as well to .gitignore. the file does not have any meaning outside local enviroments

pierotofy, 2016-09-11:
Already there, see https://github.com/OpenDroneMap/WebODM/blob/master/.gitignore#L54