WebODM/WebODM #11: Communication with node-OpenDroneMap

pierotofy, 2016-09-12:
In the next few weeks I'll dedicate some time to write the code necessary to create new `ProcessingNode` instances and have them communicate with https://github.com/pierotofy/node-OpenDroneMap. I will likely spend some time tweaking node-OpenDroneMap to take care of authorization issues and to make the process of setting up new `ProcessingNode instances easy.

After careful evaluation, I think I will not add Celery as discussed in chat to the mix; the additional complexity added by Celery does not seem to simplify much the problem of distributing tasks to multiple nodes. This is particularly true because the number of processing tasks that we can expect are not many, so I think a simpler cron-like lib, with some custom logic, a priority queue and basic error handling/retry will serve the purpose (hopefully) better.

pierotofy, 2016-09-20:
Added working proof of concept, used bravado to establish communication with node-OpenDroneMap.

Note to self: if installation fails via docker, need to run `pip install --upgrade git+git://github.com/Yelp/swagger_spec_validator`

dakotabenjamin, 2016-09-21:
I haven't had a chance to check this out yet, but I'll be able to starting next week.

pierotofy, 2016-09-21:
ProcessingNode.update_node_info() is now able to update django models with external info from node-OpenDroneMap.

I still need to work on some basic authentication mechanism.

pierotofy, 2016-09-29:
I will put authentication aside from the moment, since the idea is that processing nodes will be part of an internal network or docker swarm, which has isolated containers. It might be interesting to incorporate something like JSON Web Tokens in the future for public facing nodes.

Still need to create a cron task that updates the nodes at specific intervals (30 secs perhaps).

pierotofy, 2016-10-11:
This is working.

closed 2016-10-11 by pierotofy

smathermather, 2016-10-12:
w00t!

Bavar2142, 2016-10-12:
Itll be interesting to see how well it works should you look at public facing nodes.