WebODM/WebODM #13: Python 2 or 3?

vesnikos, 2016-09-15:


dakotabenjamin, 2016-09-15:
OpenDroneMap is in Python 2 so in the short term we should stick with that.

pierotofy, 2016-09-15:
I want badly to use Python 3, but the reason I've initially setup WebODM to use the 2.7 image is because I don't yet know for sure what libraries we are going to need (and which ones perhaps we can work around, or substitute). There seems to be quite a big of fragmentation still going on as of today, many packages simply do not work in 3.

My take was to develop using Python 2, making sure to follow Django's best practices for compatibility: https://docs.djangoproject.com/en/1.10/topics/python3/. Then as soon as we have a good idea of what libraries we are using, check if they work on Python 3, and make the switch.

@vesnikos do you foresee any problem in using Python 3 directly? I would like to take a gamble and start targeting 3 directly, I just don't know all of the requirements of the project at this point.

pierotofy, 2016-09-25:
Opened #16

smathermather, 2016-09-25:
Ah, ya. I forgot about his one.

pierotofy, 2016-11-08:
Switched to Python 3, see #16

closed 2016-11-08 by pierotofy