WebODM/WebODM #18: Install - step 13 run curl failure

dakotabenjamin, 2016-09-27:
Ran into this issue while running `docker-compose up`

```
Step 13 : RUN curl --silent --location https://deb.nodesource.com/setup_6.x | sudo bash -
 ---> Running in 6537e09cc0d5
/bin/sh: 1: sudo: not found
ERROR: Service 'webapp' failed to build: The command '/bin/sh -c curl --silent --location https://deb.nodesource.com/setup_6.x | sudo bash -' returned a non-zero code: 127
```

My specs: 
- Ubuntu 14.04 server

```
dmb2@OpenDroneMap:~/WebODM$ docker version
Client:
 Version:      1.12.0
 API version:  1.24
 Go version:   go1.6.3
 Git commit:   8eab29e
 Built:        Thu Jul 28 22:00:36 2016
 OS/Arch:      linux/amd64

Server:
 Version:      1.12.0
 API version:  1.24
 Go version:   go1.6.3
 Git commit:   8eab29e
 Built:        Thu Jul 28 22:00:36 2016
 OS/Arch:      linux/amd64
dmb2@OpenDroneMap:~/WebODM$ docker-compose --version
docker-compose version 1.8.1, build 878cff1
```

pierotofy, 2016-09-27:
Since we're running everything as root, we can probably remove the `sudo` from `| sudo bash -`.

pierotofy, 2016-09-27:
I'll make a pull request for this.

dakotabenjamin, 2016-09-27:
Great! Thanks.

closed 2016-09-27 by dakotabenjamin