WebODM/WebODM #22: Image Upload + Project/Task creation

pierotofy, 2016-09-29:
Somewhat linked to #21, there's going to be a button for uploading images directly from the dashboard. This will in turn automatically create the first project and task. These are all shortcuts for the sake of ease of use, and a more in depth project and task management UI will be created.

pierotofy, 2016-10-11:
Will tackle this next.

pierotofy, 2016-10-18:
Image upload is working, as well as task creation. Still need to add project creation API, update task info.

pierotofy, 2016-10-22:
Update task info is done.

pierotofy, 2016-11-08:
Still need UI to create new projects. API is already in place to support this.

jjmata, 2016-11-12:
This is working pretty well already, one limitation I've noticed is that a job cannot be restarted on a separate node (which sometimes would be the most desirable thing since some nodes might be better spec'ed and could handle a job that fails elsewhere)

Noting a little SQL couldn't fix (`update app_task set processing_node_id = 2 where id = 4`, need UI to support it.  If I new my React I could even PR it! ;-)

pierotofy, 2016-11-12:
Yup, there's going to be an edit button that will allow users to switch processing node / options. I haven't gotten around to write the code for it yet, some other items are taking priority at the moment.

jjmata, 2016-11-12:
Actually, spoke too soon, looks like restarting on a separate node needs more than just a little SQL hacking, got stuck in `Restarting...`:

[image]

pierotofy, 2016-11-12:
If you're going to hack it, you currently need to reset the status column (set it to null) and remove the uuid (set it also to null).

`update app_task set processing_node_id = 2, uuid = null, status = null where id = 4`

jjmata, 2016-11-12:
Thanks!  Just getting my hands dirty to understand `WebODM` and hopefully contribute.  One other thing I noticed is that you can easily navigate away from an tab uploading and kill it mid-way with no warning.  Need a Javascript warning for that.

pierotofy, 2016-11-12:
That would be a good improvement. Perhaps open an issue and assign it to yourself, or leave it for somebody to take care of it.

jjmata, 2016-11-12:
Sure, [here it is](https://github.com/OpenDroneMap/WebODM/issues/45).

pierotofy, 2016-11-17:
Projects can be created/deleted, tasks can be created/deleted. This can be closed.

closed 2016-11-17 by pierotofy