WebODM/WebODM #24: Free Processing Nodes Volunteer Network

pierotofy, 2016-10-03:
At some point it could be beneficial to create a basic infrastructure for allowing people to donate server resources to process maps.

It could be as simple as having volunteers install the processing software, having a sign-up form, and some code to check which nodes are online.

Bavar2142, 2016-10-04:
Righto. Ill have a look at boinc to see how theyve organised it and get back to you.

Bavar2142, 2016-10-05:
Boinc itself is under the ​Lesser General Public License (LGPL), version 3 or there abouts and i've managed to find links to their git repository and source code.

https://boinc.berkeley.edu/trac/wiki/SourceCodeGit

pierotofy, 2016-11-07:
This idea would require:
 - Central endpoint to maintain a list of active nodes
 - Modification to node-OpenDroneMap to add an `advertise` or `broadcast` or `public` flag that automatically notifies the central list
 - Addition of a sponsor name, other metadata to node-OpenDroneMap
 - Code that fetches the list of public nodes in WebODM

skypuppy, 2019-01-02:
Not sure if BOINC can do this automagically, but it would be ideal if we could distribute the work load over a number of systems.  Yeah, I know, distributed processing is a whole 'nother can of worms.

smathermather, 2019-01-03:
We’re working on solving some of the distributed problems anyway, it’d just be a different network to distribute on, so distribution isn’t too much of an issue, or won’t be once we’re done with it :D

pierotofy, 2020-04-02:
I'm going to take this off the TODO list. There's too many unresolved difficulties:
 - Unsustainable economic costs of running and managing such network
 - Data privacy and security concerns
 - Impossibility(?) of guaranteeing QOS 

I'd welcome proposals and coding efforts from the community to solve this.

closed 2020-04-02 by pierotofy