WebODM/WebODM #37: GCP file support

pierotofy, 2016-10-25:
I currently skipped the parts of code to handle GCP files. This will need to be added at some point.

smathermather, 2016-10-27:
@mojodna will be working on an interface interface for GCP creation in POSM soon.

spatialsteve, 2017-02-17:
FYI I am working on a quicker and easier way to create a GCP file for ODM, would this be the best place to share my progress?

pierotofy, 2017-02-17:
Absolutely! :smile:

spatialsteve, 2017-02-17:
Great! I'll be sure to post here :-)

smathermather, 2017-02-17:
Excellent! How far in are you? We do have a project in progress to do just this. Depending on where you are in the process, maybe it'd help for us to share too... .

spatialsteve, 2017-02-17:
At present it is just a concept we are working on, really just aiming initially at making it easier to create the gcp text file, as far as deep integration with the WEbODM workflow, I am not sure how the GCP are currently used. If they are used at the start of the process to increase the accuracy of all processing then what I have conceptualised would work well.

I have some ideas for the logic too but yet to fully flesh it out. This is a wireframe for reference.
[image]

smathermather, 2017-02-20:
Excellent. We'll work on releasing what we have so far as soon as possible.

smathermather, 2017-02-22:
Ok, it is now opened up, @spatialsteve:
https://github.com/stamen/gcpi

Feedback very welcome. I'd be interested to know how it compares with your vision.

spatialsteve, 2017-02-22:
Great @smathermather I'll take a look - just heading out to capture sunrise footage ;-) thanks for sharing.

byezy, 2017-04-11:
Great work guys... so is it the case now that ODM cannot utilise GCPs during processing? I would be very interested in knowing if/when this is planned, and also would happy to do some testing of this functionality as I have access to some commercial software against which to compare results...(if that would help at all). Cheers

pierotofy, 2017-04-11:
WebODM doesn't support GCPs yet, ODM and node-OpenDroneMap do.

smathermather, 2017-04-11:
@pierotofy -- when you get a chance, we have some good updates to https://github.com/stamen/gcpi .

smathermather, 2017-04-11:
Oops. looks like updates haven't been pushed yet. Disregard for the time being @pierotofy.

pierotofy, 2017-04-13:
Bummer, I was hoping to start working on this next.

smathermather, 2017-04-15:
Just bugged to have it pushed to the repo.

pierotofy, 2017-05-15:
We now have GCP support via txt file upload (thanks to @gkconsulting ), the GCP interface integration will follow.

LeneFischer, 2017-08-08:
Any news about GCP interface integration. Is there a link to read more about creating GCP ?

pierotofy, 2017-08-08:
We haven't had time to work further on the GCP interface yet, but it's quite high up in the priority list. Some links to checkout: https://github.com/OpenDroneMap/OpenDroneMap/wiki/Running-OpenDroneMap#running-odm-with-ground-control and https://smathermather.com/2015/03/01/opendronemap-gcp-file-format/

dani2007, 2017-11-09:
To creat a GCP file still has to been done manually, right? 
If anyone has a good work flow, I'd apreciate any hints. If not, I'll ceat a txt file and use paint to get the pixel row and collum ;-). It's some work for 10 GCP but it will work. 

Thanks for your wirk!

gkconsulting, 2017-11-09:
Hi, 
I have not done much work on this lately, however I have used this fork and works really well to generate my GCP file. Its a great and easy automated method.
https://github.com/wolkstein/OpenDroneMap-GCP_LIST.TXT-generator

I hope this helps

dani2007, 2017-11-09:
ah, thanks. I'll try this.

dakotabenjamin, 2017-11-09:
@gkconsulting that is awesome. cc. @smathermather

kikislater, 2017-11-09:
Whaoooh ! 
Nice feature

dani2007, 2017-11-10:
okey, I have no Idea about python or linux  ;-P. Well I was proud already, that I got webODM to run ;-)

Well now I did al the steps with the the tempate images and so on. but how do I start the skript on windows? ;-). sorry for the stubid querstion

uasimagery, 2017-11-10:
Start docker then:
cd webodm

then:
webodm.sh start

dani2007, 2017-11-10:
Sorry, no that was not my question. that I did get ;-)

I am trying to run the "https://github.com/wolkstein/OpenDroneMap-GCP_LIST.TXT-generator"

dani2007, 2017-11-11:
okey, I spent almoust a full day now to try to get the script running on a windows 10 mschine. 

at first I tried it with python 3, witch did not seam to work. than I figured maybe I need python 2, then I tried to install numpy, in many different ways, with somehow did not work. than I installed the Miniconda package, So i could run nimpy. Then I had to install CV2 somehow. and now gives me this:

(C:\Users\seich\AppData\Local\conda\conda\envs\my_root) C:\Users\seich>python C:\Bachserei\OpenDroneMapGCP\gcp_txt_gen.py
Found 0 Images
---sorted tmpImagePathList
Found 0 Images-Templates
Traceback (most recent call last):
  File "C:\Bachserei\OpenDroneMapGCP\gcp_txt_gen.py", line 121, in <module>
    sift = cv2.SIFT()
AttributeError: 'module' object has no attribute 'SIFT'

(C:\Users\seich\AppData\Local\conda\conda\envs\my_root) C:\Users\seich>

dani2007, 2017-11-11:
Okey, I tinkerded around a liitle more. i found out, that SIFT is no longer in openCV. :-(

https://www.pyimagesearch.com/2015/07/16/where-did-sift-and-surf-go-in-opencv-3/

pierotofy, 2018-04-19:
We know have the `posm-gcpi` plugin, so this can be probably closed.

closed 2018-04-19 by pierotofy