WebODM/WebODM #41: Should Celery be used here?

pierotofy, 2016-11-03:
In regard to a discussion which began here: https://github.com/OpenDroneMap/WebODM/pull/40#pullrequestreview-7092904

Is `scheduler.py` code reinventing the wheel (functionality that already exists in Celery)?

In essence it [the code] is replicating some of the Celery features (it is a task processing queue). What is driving me to re-implement this mechanism is that so far, the task processing involves more than a single processing step (think input images in ---> assets out). Maybe I'm over-thinking it, but so far I found these cases where Celery does not seem like it would fare so well.

Input images in --> assets are not what I expected. I should be able to restart the process without re-sending the images, just by changing the processing options?

Console output: how to retrieve real-time console output from the opendronemap process?

How to protect processing nodes (authentication) without making it difficult to setup? Ideally a processing node should be (optionally) protected by a static token, or a userID/password combination that generates a temporary token. The docs here hint at using public/private key exchange for security: http://docs.celeryproject.org/en/latest/userguide/security.html (I'm not sure I'd want to go down that route)

pierotofy, 2016-11-07:
For 3. I could send tokens as part of the messages, and do my own authentication.

Multi-step processes (1, 2) are still an issue.

mojodna, 2016-11-07:
Chords, groups, and chains facilitate multi-step processes (and ones that require coordination), e.g.: https://github.com/AmericanRedCross/posm-imagery-api/blob/master/app.py#L110-L119

Aside: this might actually be a more appropriate way of running tasks in ODM core than Ecto, even when not run in a "task" context.

For (1), I prepare a "project directory" (kind of) and mount key paths into an ODM Docker image (though it could just as easily point to those paths w/o the containerization). That would be step 1, step 2 would be the first stage that begins to manipulate data.

For (2), I pipe output to files within a directory that's generally accessible (shared filesystem between the host and container) and expose it through an API endpoint: https://github.com/AmericanRedCross/posm-opendronemap-api/blob/master/app.py#L362-L379

Re: (3), I'm thinking about it the other way. Processing nodes needn't be publicly-addressable. Instead, they would opt into processing by subscribing to "the" (or one of) the topics Celery uses (either RabbitMQ or Redis) to coordinate tasks. At that point, protecting the broker becomes a single place to limit access: if a worker can connect to the broker, it's assumed to be able to process tasks.

pierotofy, 2016-11-07:
I don't think I have enough experience using Celery to understand how 1) would be solved. Wouldn't that require a new transfer of all the images? I guess I'd love for somebody to help me out with this.

2) Seems like a good workaround.

3) Mm yes, I guess it depends on the use cases; for an organization it would make a lot of sense. For setting up something like #24 it wouldn't work as nicely. Supporting both would probably be best in the long term.

mojodna, 2016-11-21:
(1) yes, but I think that's the case whenever processing is intended to occur somewhere other than the initial ingestion location.

(3) there might be a way to handle it, but agreed, not using the current regime. (I like the idea of drive-by contributors of CPU time, but I'll admit to being a bit skeptical about it, specifically around the constraints it applies to the core architecture.)

pierotofy, 2016-11-21:
For 1), node-OpenDroneMap will keep the images in storage until they are either deleted explicitly or until the node has decided to purge them (after 3 days by default), so a restart of the process can happen without reuploading the files. Not sure if this is a big advantage, probably not, but it's there.

I can see problems going down the road using my current approach (the same problems that a task processing queue library like Celery has already solved), so I'd want to migrate to it.

Perhaps when there's some time available somebody could help me with this.

jjmata, 2016-11-23:
> so a restart of the process can happen without reuploading the files

I actually believe that a restart of the process is a great advantage when we are talking of large datasets such as these.  I can see myself submitting multiple tasks based on the same dataset for experimentation (and even to fix issues discovered on different versions of ODM).

pierotofy, 2016-11-23:
The user can restart the process using either approach (WebODM never deletes user uploads), but it's a matter of how WebODM then uploads the images for processing to a node.

User --> WebODM *--->* processing node.

The bold arrow is the part that we are discussing. In a dedicated infrastructure (WebODM and processing nodes are in the same network), the overhead is little, across the internet or on slow connections, overhead adds a bit. But this is important only when it comes to restart a process.