WebODM/WebODM #42: Upload Issue

pierotofy, 2016-11-09:
if the upload is quicker than you are at providing a name / clicking "Save" then you are stuck where you cannot submit the job anymore ...

pierotofy, 2016-11-09:
@jjmata_twitter I've tried to reproduce this, but I'm having some troubles... the only time where I can get the "Processing node is not set" message is if I upload some images, and then refresh the page.

Would you have a list of reproduction steps I could try to follow?

jjmata, 2016-11-09:
Sure, wasn't doing anything special other than maybe using Safari.  I'm in the middle of taking today's changes in for a spin ... will tell you if I can reproduce.

jjmata, 2016-11-10:
Seems to happen all the time to me on Safari Version 10.0.1 (12602.2.14.0.7), where the task just never gets the metadata persisted.  Chrome works fine however.

pierotofy, 2016-11-10:
That would make sense; I'll do some troubleshooting with Safari. Thanks a lot for the info!

jjmata, 2016-11-11:
OK, when using Safari, the `PATCH` which comes after the `POST` that creates it, and updates your task definition with name / config options / processing nodes never gets to the server:

```
webodm                | INFO "POST /api/projects/1/tasks/ HTTP/1.1" 201 9
webodm                | INFO "GET / HTTP/1.1" 302 0
```

But it does on Chrome:

```
webodm                | INFO "POST /api/projects/1/tasks/ HTTP/1.1" 201 9
webodm                | INFO "GET /api/projects/1/tasks/13/ HTTP/1.1" 200 299
webodm                | INFO "GET /api/projects/1/tasks/13/ HTTP/1.1" 200 299
webodm                | INFO "PATCH /api/projects/1/tasks/15/ HTTP/1.1" 200 262
```

pierotofy, 2016-11-11:
I wrote a fix for this in the `pierotofy/actions` branch, which will land in master on the next pull request once it finishes review.

Problem was caused by lack of Object.values() in Safari (and IE and Opera), which is an experimental feature . Problem is solved using a polyfill.

commit 8eb884d7 (2016-11-15): Fixed #42
closed 2016-11-11 by pierotofy