WebODM/WebODM #45: Stop a user from mindlessly navigating away from an image upload session

jjmata, 2016-11-12:
Throw some sort of JS error to confirm, and clean up the mess (since it's probably easier than trying to implement "re-start upload" for now).

Any React pointers for me on how this could be done, @pierotofy?  Happy to take it from there ...

pierotofy, 2016-11-12:
I'd read the react tutorial here https://facebook.github.io/react/tutorial/tutorial.html, it's not too long, then you'd be looking at changing the logic in `app/static/app/js/components/ProjectListItem.jsx`. Check for the browser back button event, if `state.uploading` is true, then prompt a confirm?

jjmata, 2016-11-12:
Got it!

jjmata, 2016-11-12:
Feel free to assign to me @pierotofy, can't seem to do it myself ...

jjmata, 2016-11-14:
I tried this approach from [here](https://github.com/ReactTraining/react-router/blob/v2.4.0/upgrade-guides/v2.4.0.md) but `props` is always `undefined` and I'm too early in my React learning to debug too well:

```
import React from 'react'
import { withRouter } from 'react-router'

const Page = React.createClass({

  componentDidMount() {
    this.props.router.setRouteLeaveHook(this.props.route, () => {
      if (this.state.unsaved)
        return 'You have unsaved information, are you sure you want to leave this page?'
    })
  }

  render() {
    return <div>Stuff</div>
  }

})

export default withRouter(Page)
```

Adding @apercas who can probably do this much quicker for us.

pierotofy, 2016-11-15:
I'm not sure you'll need react-router. Also note that you can use ES6 style classes!

pierotofy, 2016-12-14:
Any news on this one? :)

jjmata, 2016-12-15:
I passed it on to @apercas but he just finished getting going on our environment: new hardware took forever to arrive, environment issues.  I'll let him comment ... but since he's the React person in my team, there is no sense in me doing it since I don't have time right now to pick up new stuff.  I wish!

apercas, 2016-12-20:
Hi, sorry I've been out of reach but fully operative again. I'll come back with news ASAP.

pierotofy, 2017-02-08:
There's a fix for this in the `gui` branch which will be merged soon into master.

commit e805a9d7 (2017-02-12): Fixes #45
closed 2017-02-08 by pierotofy