WebODM/WebODM #51: Processing error

boxerone, 2016-11-25:
Hi there
I am not sure if this error is related to webODM or the processig node

My setup is two different AWS instance (t2.medium) one running WebODM and the second one running Node-ODM, both installed with Dockers

```
Killed
[ERROR]   quitting cause: 
	/code/SuperBuild/install/bin/pmvs2 /var/www/data/813e7f99-16a8-4ec7-b5d8-edaa958957aa/pmvs/recon0/ option-0000
returned with code 35072.
```

any help will be very helpful

pierotofy, 2016-11-25:
Hi, thanks for reporting the issue.

It might be related to this: https://github.com/OpenDroneMap/OpenDroneMap/issues/28

Possible causes: 
 - Dataset is too large (or AWS instance doesn't have sufficient memory)
 - The images don't have sufficient overlap
 - The images are too small

Perhaps you could provide us with more details about the dataset you are trying to process?

boxerone, 2016-11-26:
Ciao Piero
reading through that post is seems that the memory might be the problem. I am running a small instance just for testing purpose but I will definitly try with a "bigger" one.
I am processing one of the comunity dataset:
https://github.com/OpenDroneMap/odm_data_bellus/

I will try again and let you know.
thansk
adriano

pierotofy, 2016-11-26:
If you are using a small instance, try to enable a swap file if you don't want to spend for a more memory: https://www.digitalocean.com/community/tutorials/how-to-add-swap-space-on-ubuntu-16-04

pierotofy, 2016-11-27:
By the way, I'm also having the same error processing that dataset (my processing node uses 4 gb ram). I'm investigating right now...

boxerone, 2016-11-27:
Correct same as mine.
Let me try increasing ram...

pierotofy, 2016-11-27:
I can confirm this is a memory issue; when I resize the images to 1800 using the "--resize-to 1800" option, processing continues without the error.

boxerone, 2016-11-28:
Great!
I've just switched to a m4.xlarge instace and processing is running without problems.

Quite frankly it seems that now is stucked at 60%, "Generating Overview Tiles"....
[image]

pierotofy, 2016-11-28:
Could be related to #55 

If you get the UUID of the task assigned by node-OpenDroneMap (look for /var/www/data/<UUID>/ in the console output), take the UUID and the URL of the node-OpenDroneMap API (for example, http://nodeodm.masseranolabs.com:80/task/UUID/download/all.zip ).

If the download starts, this is likely the same issue as #55.

Also in the output of WebODM (from `python manage.py runserver`) if you see:

```
INFO Acquiring lock: Task ID: xx
INFO Processing status: 40 for Task ID: xx
```

This is probably also the same as #55.

boxerone, 2016-11-29:
So, processing the same images dataset, from the web interface of the node-odm, provides rusults with no problem.
Reprocessing from Web-odm, got the same error

pierotofy, 2016-12-01:
I will investigate this and get back to you.

pierotofy, 2017-01-15:
Sorry for the late reply! This issue should be now solved as part of the fix for #55. I'll close this for now, unless the problem continues.

closed 2017-01-15 by pierotofy