WebODM/WebODM #54: Elevation Model

pierotofy, 2016-11-26:
It would be cool to have the ability to display an automatically generated elevation model along with the orthophoto.

@smathermather how would you go about generating an elevation model from the current outputs of opendronemap (lastools come to mind, but they aren't OSS)?

dakotabenjamin, 2016-11-28:
Probably PDAL would be the way to go, since we've already got that package in ODM

pierotofy, 2016-11-28:
Oh nice, I did not know PDAL supported that. Found this after more digging: http://www.pdal.io/workshop/exercises/analysis/dtm/dtm.html

smathermather, 2016-11-28:
I haven't finished documenting:

https://smathermather.com/2016/10/21/filtering-buildings-and-vegetation-from-drone-derived-surface-models/

You can use PDAL to filter with vector data (e.g. building footprints) and also shape-based filtering or both combined.

smathermather, 2016-11-28:
The trick is getting PDAL installed with all the appropriate dependencies (easy with docker, difficult otherwise).

pierotofy, 2016-11-28:
@smathermather know what you mean. I built it time ago on Windows!

andrespias, 2016-12-04:
I want to generate a only-vegetation map, applying some kind of filter on the DSM generated by ODM.
ODM, generates a point cloud as a ply file (pmvs/recon0/models/option-0000.ply) right?
I've been trying to run something like this with no results:
`docker run -v /srv/torreap:/datos pdal/pdal:1.3 pdal ground /datos/odm_data/caliterra/pmvs/recon0/models/option-0000.ply -o /datos/tierra.ply --filters.ground.classify=true`
How can I convert the ODM output format into a allowed format by PDAL?

pierotofy, 2016-12-05:
@andrespias could you please open a separate issue in the OpenDroneMap repository? This isn't related to WebODM. https://github.com/OpenDroneMap/OpenDroneMap/issues

andrespias, 2016-12-18:
@pierotofy ok i will do that.

smathermather, 2016-12-21:
I'm thinking of working on this and full PDAL integration (ODM, not WebODM, but nonetheless relevant) at the code sprint in February.

pierotofy, 2018-01-30:
This has been completed a while ago.

closed 2018-01-30 by pierotofy