WebODM/WebODM #55: Task stuck on running

pierotofy, 2016-11-27:
Processing the odm_data_bellus dataset with --resize-to 1800 option, the task remains stuck on running. node-OpenDroneMap successfully processed the images.

dakotabenjamin, 2016-12-06:
I think I got a similar issue. I started this yesterday at noon, it's now been 18 hours. The console output hasn't changed since 4pm yesterday
[image]

pierotofy, 2016-12-06:
Yeah sounds like the same issue. I'm hoping to find some time to look at this problem sometimes this week. node-OpenDroneMap has finished the process (you can verify that by taking the UUID assigned to the task and manually calling the API), but there's some error in WebODM while retrieving the assets that is getting silently caught and ignored. Will post an update when I know more.

pierotofy, 2016-12-06:
And yes, the `90...100 - done` output is the tiling process (it's a call to gdal2tiles.py)

dakotabenjamin, 2016-12-06:
FWIW gdal2tiles.py was not running when examining current processes using `top`. Also this was a large dataset (500mb orthophoto from almost 600 photos)

desireB, 2016-12-07:
Hi,
I got the same error (stuck on running). I will try with a a very small set to see.

Thank you.

desireB, 2016-12-07:
I used a small set of 87 photos, and it worked great.
Great job guys. That's really cool.
Thanks.

pierotofy, 2016-12-08:
The issue is caused by the following stack trace:

```
ERROR Uncaught error: invalid memory alloc request size 1073741824
 Traceback (most recent call last):
  File "/home/piero/.envs/webodm/lib/python3.5/site-packages/django/db/backends/utils.py", line 64, in execute
    return self.cursor.execute(sql, params)
psycopg2.InternalError: invalid memory alloc request size 1073741824

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/piero/Documents/WebODM/app/scheduler.py", line 83, in process
    task.process()
  File "/home/piero/Documents/WebODM/app/models.py", line 326, in process
    self.save()
  File "/home/piero/Documents/WebODM/app/models.py", line 154, in save
    super(Task, self).save(*args, **kwargs)
  File "/home/piero/.envs/webodm/lib/python3.5/site-packages/django/db/models/base.py", line 796, in save
    force_update=force_update, update_fields=update_fields)
  File "/home/piero/.envs/webodm/lib/python3.5/site-packages/django/db/models/base.py", line 824, in save_base
    updated = self._save_table(raw, cls, force_insert, force_update, using, update_fields)
  File "/home/piero/.envs/webodm/lib/python3.5/site-packages/django/db/models/base.py", line 889, in _save_table
    forced_update)
  File "/home/piero/.envs/webodm/lib/python3.5/site-packages/django/db/models/base.py", line 939, in _do_update
    return filtered._update(values) > 0
  File "/home/piero/.envs/webodm/lib/python3.5/site-packages/django/db/models/query.py", line 652, in _update
    return query.get_compiler(self.db).execute_sql(CURSOR)
  File "/home/piero/.envs/webodm/lib/python3.5/site-packages/django/db/models/sql/compiler.py", line 1148, in execute_sql
    cursor = super(SQLUpdateCompiler, self).execute_sql(result_type)
  File "/home/piero/.envs/webodm/lib/python3.5/site-packages/django/db/models/sql/compiler.py", line 835, in execute_sql
    cursor.execute(sql, params)
  File "/home/piero/.envs/webodm/lib/python3.5/site-packages/django/db/backends/utils.py", line 79, in execute
    return super(CursorDebugWrapper, self).execute(sql, params)
  File "/home/piero/.envs/webodm/lib/python3.5/site-packages/django/db/backends/utils.py", line 64, in execute
    return self.cursor.execute(sql, params)
  File "/home/piero/.envs/webodm/lib/python3.5/site-packages/django/db/utils.py", line 94, in __exit__
    six.reraise(dj_exc_type, dj_exc_value, traceback)
  File "/home/piero/.envs/webodm/lib/python3.5/site-packages/django/utils/six.py", line 685, in reraise
    raise value.with_traceback(tb)
  File "/home/piero/.envs/webodm/lib/python3.5/site-packages/django/db/backends/utils.py", line 64, in execute
    return self.cursor.execute(sql, params)
django.db.utils.InternalError: invalid memory alloc request size 1073741824
```

It seems to affect these lines:

```python
 # Add to database orthophoto
# ....
    self.orthophoto = GDALRaster(orthophoto_path, write=True)

self.save()
```

Seems like saving a large raster does not cope well with postgis. I'll explore options.

pierotofy, 2016-12-09:
A further task stuck situation can also be triggered by the following: https://github.com/pierotofy/node-OpenDroneMap/issues/4

pierotofy, 2016-12-09:
Problem is caused by the fact that GDALRaster loads the file in memory (*gulp*) before writing it to the database. This works fine for files less than 20mb, but anything larger is bad. A file of 150mb (like the one from bellus) will push the memory usage past 1g (I'm thinking due to byte --> string encoding that is used to generate the query by the Django ORM), which throws the error.

PostGIS supports out-of-db rasters, I'm just not sure how to make it play nice with Django... worst case we just store a path and provide methods to initialize GDALRaster manually.

mojodna, 2016-12-09:
Out-of-db rasters appear to clients exactly the same as ones that are stored in the database. To enable them in PostGIS, you probably want to set the following when starting Postgres:

```bash
export POSTGIS_ENABLE_OUTDB_RASTERS=1
export POSTGIS_GDAL_ENABLED_DRIVERS=ENABLE_ALL
```

PostGIS raster is pretty awesome, but it also has some sinkholes to watch out for. You can tile up the raster when importing (the docs suggest(ed) 100x100px chunks, but that's unnecessary), in which case you end up with a row for each tile. If you load it without tiling, the raster ends up in a single row.

Last I checked, raster operations don't read subsets of external files (this is definitely true when 1 row == 1 file), so if your rasters are big, GDAL is going to be dealing with them in their entirety, even if you're only operating on smaller areas.

pierotofy, 2016-12-09:
Out-of-db would be perfect for the needs; @mojodna do you know by chance a way to import raster files as out-of-db without invoking raster2pgsql?

```
# raster2pgsql -e -R -a test.tif public.test
INSERT INTO "public"."test" ("rast") VALUES ('0100000400492BF78A1A99A93F780DE9452A99A9BF95F1EFF3EAE51A410AF4893C6D69514100000000000000000000000000000000697F000055291B2E840000746573742E74696600840001746573742E74696600840002746573742E74696600840003746573742E74696600'::raster);
```

0100000400492.....74696600 is a hex Well Known Binary string, and I found this http://postgis.net/docs/doxygen/2.2/d6/d41/raster2pgsql_8py_source.html which could help me generate the proper WKB, but I was wondering if there was an easier way. I could always do a regex to capture the output of raster2pgsql, but that seems a little too hackish.

mojodna, 2016-12-09:
Yup, using `ST_BandMetadata`. There's an example in [`ST_AddBand`](http://postgis.net/docs/RT_ST_AddBand.html#idm18968) that uses `ST_MakeEmptyRaster` to define the metadata (dimensions, upper left, scale, skew, and SRID). (That's the hex WKB you're seeing, which will differ by raster.)

pierotofy, 2016-12-09:
That seems like it should work, but I can't get past this:

```
SELECT
	ST_AddBand(
		ST_MakeEmptyRaster(2276, 2271, 576659.599854, 5188215.73153, 0.04997872539539371, -0.049993623954373956, 0, 0, 32615),
		'/home/piero/Documents/WebODM/test2.tif'::text, NULL::int[]
	);
```

```
ERROR:  RASTER_addBandOutDB: Could not open out-db file with GDAL
CONTEXT:  SQL function "st_addband" statement 1

********** Error **********

ERROR: RASTER_addBandOutDB: Could not open out-db file with GDAL
SQL state: XX000
Context: SQL function "st_addband" statement 1
```

```
SHOW postgis.enable_outdb_rasters; --> 'on'
SHOW postgis.gdal_enabled_drivers; --> "ENABLE ALL"

SELECT postgis_version(); --> "2.3 USE_GEOS=1 USE_PROJ=1 USE_STATS=1"
```

Test tif attached. Mm...

[test2.zip](https://github.com/OpenDroneMap/WebODM/files/642409/test2.zip)

pierotofy, 2016-12-09:
To make the problem more difficult, RasterField in GeoDjango always serializes to a GDALRaster object, which does not have support for out-of-db files.

Is there a place where the hex WKB format is documented? I'd make my journey a little easier than reading through all of raster2pgsql code...

pierotofy, 2016-12-09:
Found nothing, so source code it is: http://postgis.refractions.net/documentation/postgis-doxygen/df/d4b/rt__api_8c_1d055ffef592f0c45a99a81184ba42d1.html#1d055ffef592f0c45a99a81184ba42d1

pierotofy, 2016-12-09:
PostGIS HEX WKB, for future reference:

| Description | # Bytes | Value | Reference |
| --------------- | ------------------- | ------- | -------------- |
| Little Endian? | 1 | 0 or 1 | |
| Zeros | 2 | Always 0 | |
| Num Bands | 2 |  | [rt_raster_serialized_t](http://postgis.refractions.net/documentation/postgis-doxygen/d8/d5e/structrt__raster__serialized__t.html) |
| scaleX | 8 |  | [rt_raster_serialized_t](http://postgis.refractions.net/documentation/postgis-doxygen/d8/d5e/structrt__raster__serialized__t.html) |
| scaleY | 8 |  | [rt_raster_serialized_t](http://postgis.refractions.net/documentation/postgis-doxygen/d8/d5e/structrt__raster__serialized__t.html) |
| ipX | 8 |  | [rt_raster_serialized_t](http://postgis.refractions.net/documentation/postgis-doxygen/d8/d5e/structrt__raster__serialized__t.html) |
| ipY | 8 |  | [rt_raster_serialized_t](http://postgis.refractions.net/documentation/postgis-doxygen/d8/d5e/structrt__raster__serialized__t.html) |
| skewX | 8 |  | [rt_raster_serialized_t](http://postgis.refractions.net/documentation/postgis-doxygen/d8/d5e/structrt__raster__serialized__t.html) |
| skewY | 8 |  | [rt_raster_serialized_t](http://postgis.refractions.net/documentation/postgis-doxygen/d8/d5e/structrt__raster__serialized__t.html) |
| srid | 4 |  | [rt_raster_serialized_t](http://postgis.refractions.net/documentation/postgis-doxygen/d8/d5e/structrt__raster__serialized__t.html) |
| width | 2 |  | [rt_raster_serialized_t](http://postgis.refractions.net/documentation/postgis-doxygen/d8/d5e/structrt__raster__serialized__t.html) |
| height | 2 |  | [rt_raster_serialized_t](http://postgis.refractions.net/documentation/postgis-doxygen/d8/d5e/structrt__raster__serialized__t.html) |
| pixtype | 1 | rt_pixtype value bit-wise ORred optionally with BANDTYPE_FLAG_OFFDB (1<<7), BANDTYPE_FLAG_HASNODATA (1<<6) and BANDTYPE_FLAG_ISNODATA  (1<<5) | [rt_pixtype](http://postgis.refractions.net/documentation/postgis-doxygen/d7/d68/rt__api_8h_43b88be0a4efc9d67d5c69099fc7776c.html#43b88be0a4efc9d67d5c69099fc7776c) |
| NoData | 1..8 | Size depends on pixtype | |
| Data/Path | 1..EOF | If pixtype contains the BANDTYPE_FLAG_OFFDB flag, see "Path" definition, otherwise "Data" definition | |

Path:

| Description | # Bytes | Value | Reference |
| --------------- | -----------| ------- | -------------- |
| Band Num | 1 | | |
| Path | 1..EOF | File path of the out-of-db raster file | |

Data:

| Description | # Bytes | Value | Reference |
| --------------- | ------------------- | ------- | -------------- |
| Data | 1..EOF | Raster Data | |

mojodna, 2016-12-09:
Check the permissions on that file / try using one in `/tmp` just to check. It may be that postgres can't read under your home directory.

The Django part shouldn't matter; when reading data from pgraster, it comes back as though it was stored in the db. The only place that distinction matters is when adding rows.

pierotofy, 2016-12-09:
Moved file to /tmp, chmod 777 on it, still same error.

As for the Django part, things have to be modified because [RasterField](https://docs.djangoproject.com/en/1.10/ref/contrib/gis/model-api/#rasterfield) currently is not able to deserialize an out-of-db raster. As soon as I can serialize/deserialize the HEXWKB information, I'm going to write a new field model that extends RasterField to handle the orthophoto.

mojodna, 2016-12-09:
Huh. I'm out of ideas. `raster2pgsql` does work?

pierotofy, 2016-12-09:
Yeah raster2pgsql works great. Maybe it's my machine. In any case, I was going to use ST_AddBand as a way to avoid doing the HEXWKB serialization/deserialization, but it looks like I'll need to do it anyways (it's a requirement of Django fields to be able to go both ways, ST_AddBand only goes one way).

pierotofy, 2016-12-13:
This is fixed as soon as #66 is merged in, which adds offdb raster support to Django.

[image]

PR #66 (2016-12-14): Task stuck fix
This is an update to fix #55, #63 and start #48. It also contains minor CSS fixes and better error handling during processing.

@mojodna any feedback is greatly appreciated! As you can see I added support for offdb rasters in Django via a new "OffdbRasterField", which replaces the classical "RasterField".

I plan to merge in these changes by Thursday.
closed 2016-12-13 by pierotofy