WebODM/WebODM #58: adding a processing node , from node-OpenDroneMap

desireB, 2016-12-01:
I started a processing node using https://github.com/pierotofy/node-OpenDroneMap .
I can access it at 127.0.0.1:3000 
I started WebODM at 127.0.0.1:8000 and added a processing node (127.0.0.1 , port 3000). WebODM tells me that my processing node seems to be offline.

pierotofy, 2016-12-01:
Hi, thank you for reporting the issue.

If you add `nodeodm.masseranolabs.com:80` as a processing node, does WebODM recognize it?

pierotofy, 2016-12-01:
Note that you cannot use `127.0.0.1` as the IP address (I will add a note in the Readme), since that will point to the dockerized container (the container running WebODM itself), you need to use the IP of your computer's network interface. On Linux you can use the `ifconfig` command to find that out, on Windows you need to go to the Network and Sharing Center, on macOS you need to look up your Network Settings.

Hope this makes sense.

Romi-Arg, 2016-12-01:
Hi Piero. If I add nodeodm.masseranolabs.com:80 as a processing node, WebODM does recognize it. However, I can't manage to set up my own processing node after following all the steps (commands) in https://github.com/pierotofy/node-OpenDroneMap and the readme file. I'm trying setting it up with my computer's IP but it also seems to be "offline".

desireB, 2016-12-01:
Hi Piero. Thank you very much !! I used the IP address on my local network , 10.0.x.x and it's working.
@Romi-Arg , you used your router (DHCP/static) provided IP ?
I did try with nodeodm.masseranolabs.com:80 , and I got an error after some time .
`
Harris running ...2974 harris done
DoG running...4253 dog done
done
Killed
[ERROR]   quitting cause: 
	/code/SuperBuild/install/bin/pmvs2 /var/www/data/ebfcf4c9-be8d-47fd-b444-2f755d4bd030/pmvs/recon0/ option-0000
returned with code 35072.

An error occurred. Check stdout above or the logs.
Killed
[ERROR]   quitting cause: 
	/code/SuperBuild/install/bin/pmvs2 /var/www/data/ebfcf4c9-be8d-47fd-b444-2f755d4bd030/pmvs/recon0/ option-0000
returned with code 35072.

An error occurred. Check stdout above or the logs.
`
But that's another issue.
You can close this and mark it as solved.

desireB, 2016-12-01:
I'm using the seneca dataset from OpenDroneMap.

closed 2016-12-01 by desireB

dakotabenjamin, 2016-12-01:
It's probably a memory error. If you can't increase the memory on your machine, you can try reducing `--cmvs-maxImages` to 250 or smaller. It will use less memory but take longer (more clusters to process)

Romi-Arg, 2016-12-01:
Working now using local Ip for seting up the processing node. Thanks!