WebODM/WebODM #61: Newbie Help

macsee, 2016-12-04:
Hi! Such a great work! I'm very interested in the project, however I can't managed to make it work yet. When I launch the web interface and add a task, system says "There are no processing nodes available. Make sure at least one of them is reachable.". How can I create a processing node? I thought this was already included with this interface, but I was mistaken. Is there any tutorial available or walktrough process from the image loading to the final orthomosaic? A little help would be nice. Thanks in advance! Cheers!

pierotofy, 2016-12-04:
Hi,

Would you be able to tell us how you built / launched WebODM (what commands did you type).

Also make sure to use the latest version. From the WebODM directory:

```
git pull origin master
```

macsee, 2016-12-04:
Hi. Thanks for answering!
I pulled from the repository a few days ago, and I'm using it from OS X using docker, along with the commands posted in this github:

```
git clone https://github.com/OpenDroneMap/WebODM
cd WebODM
pip install docker-compose
docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml up
```

I can't find where is the "node-odm-1" processing node. How can I create more processing nodes?
Thanks!

pierotofy, 2016-12-04:
Try to run docker-compose with the `--force-recreate` param, perhaps the images you are using are not current.

`docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml up --force-recreate`

This should only be needed once.

For setting up new processing nodes, see the instructions posted here: https://github.com/pierotofy/node-OpenDroneMap

macsee, 2016-12-04:
Ok. I'll try it and let you know.
Thank you.
Cheers!

macsee, 2016-12-04:
After a long downloading I executed the commands you told me and I've got this error. Now the web interface is not even showing. Don't know what to do. Could you help me here please! Thanks!

``
node-odm-1    | info: Initialized 0 tasks
node-odm-1    | info: Checking for orphaned directories to be removed...
node-odm-1    | info: Server has started on port 3000
db            | LOG:  database system was shut down at 2016-12-04 22:02:44 UTC
webapp        | wait-for-it.sh: waiting 15 seconds for db:5432
db            | LOG:  MultiXact member wraparound protections are now enabled
webapp        | wait-for-it.sh: db:5432 is available after 0 seconds
webapp        | 
webapp        |  _       __     __    ____  ____  __  ___
webapp        | | |     / /__  / /_  / __ \/ __ \/  |/  /
webapp        | | | /| / / _ \/ __ \/ / / / / / / /|_/ / 
webapp        | | |/ |/ /  __/ /_/ / /_/ / /_/ / /  / /  
webapp        | |__/|__/\___/_.___/\____/_____/_/  /_/   
webapp        | 
webapp        | 
db            | LOG:  database system is ready to accept connections
db            | LOG:  autovacuum launcher started
db            | LOG:  incomplete startup packet
webapp        | Checking python version... 3.x, good!
webapp        | Building asssets...
webapp        | Hash: 0120632953aad14a60c2
webapp        | Version: webpack 1.13.3
webapp        | Time: 7057ms
webapp        |                        Asset     Size  Chunks             Chunk Names
webapp        | main-0120632953aad14a60c2.js  1.52 MB       0  [emitted]  main
webapp        |                 css/main.css  30.1 kB       0  [emitted]  main
webapp        |    [0] multi main 28 bytes {0} [built]
webapp        |     + 254 hidden modules
webapp        | 
webapp        | ERROR in ./app/static/app/js/Console.jsx
webapp        | Module not found: Error: Cannot resolve module 'immutability-helper' in /webodm/app/static/app/js
webapp        |  @ ./app/static/app/js/Console.jsx 19:26-56
webapp        | 
webapp        | ERROR in ./app/static/app/js/components/Map.jsx
webapp        | Module not found: Error: Cannot resolve module 'leaflet-measure/dist/leaflet-measure.css' in /webodm/app/static/app/js/components
webapp        |  @ ./app/static/app/js/components/Map.jsx 25:0-51
webapp        | 
webapp        | ERROR in ./app/static/app/js/components/Map.jsx
webapp        | Module not found: Error: Cannot resolve module 'leaflet-measure/dist/leaflet-measure' in /webodm/app/static/app/js/components
webapp        |  @ ./app/static/app/js/components/Map.jsx 27:0-47
webapp        | 
webapp        | ERROR in ./app/static/app/js/components/Paginated.jsx
webapp        | Module not found: Error: Cannot resolve module 'immutability-helper' in /webodm/app/static/app/js/components
webapp        |  @ ./app/static/app/js/components/Paginated.jsx 13:26-56
webapp        | 
webapp        | ERROR in ./app/static/app/js/components/ProjectListItem.jsx
webapp        | Module not found: Error: Cannot resolve module 'immutability-helper' in /webodm/app/static/app/js/components
webapp        |  @ ./app/static/app/js/components/ProjectListItem.jsx 15:26-56
webapp        | Child extract-text-webpack-plugin:
webapp        |         + 2 hidden modules
webapp        | Child extract-text-webpack-plugin:
webapp        |         + 2 hidden modules
webapp        | Child extract-text-webpack-plugin:
webapp        |         + 2 hidden modules
webapp        | Child extract-text-webpack-plugin:
webapp        |         + 2 hidden modules
webapp        | Child extract-text-webpack-plugin:
webapp        |         + 2 hidden modules
webapp        | Child extract-text-webpack-plugin:
webapp        |         + 2 hidden modules
webapp        | Child extract-text-webpack-plugin:
webapp        |         + 2 hidden modules
webapp        | Child extract-text-webpack-plugin:
webapp        |         + 2 hidden modules
webapp        | Child extract-text-webpack-plugin:
webapp        |         + 2 hidden modules
webapp        | Child extract-text-webpack-plugin:
[... 73 lines cut ...]
webapp        | Unhandled exception in thread started by <function check_errors.<locals>.wrapper at 0x7fda63da9ea0>
webapp        | Traceback (most recent call last):
webapp        |   File "/usr/local/lib/python3.5/site-packages/django/utils/autoreload.py", line 226, in wrapper
webapp        |     fn(*args, **kwargs)
webapp        |   File "/usr/local/lib/python3.5/site-packages/django/core/management/commands/runserver.py", line 113, in inner_run
webapp        |     autoreload.raise_last_exception()
webapp        |   File "/usr/local/lib/python3.5/site-packages/django/utils/autoreload.py", line 249, in raise_last_exception
webapp        |     six.reraise(*_exception)
webapp        |   File "/usr/local/lib/python3.5/site-packages/django/utils/six.py", line 685, in reraise
webapp        |     raise value.with_traceback(tb)
webapp        |   File "/usr/local/lib/python3.5/site-packages/django/utils/autoreload.py", line 226, in wrapper
webapp        |     fn(*args, **kwargs)
webapp        |   File "/usr/local/lib/python3.5/site-packages/django/__init__.py", line 27, in setup
webapp        |     apps.populate(settings.INSTALLED_APPS)
webapp        |   File "/usr/local/lib/python3.5/site-packages/django/apps/registry.py", line 85, in populate
webapp        |     app_config = AppConfig.create(entry)
webapp        |   File "/usr/local/lib/python3.5/site-packages/django/apps/config.py", line 90, in create
[... cut ...]
webapp        |   File "/usr/local/lib/python3.5/importlib/__init__.py", line 126, in import_module
webapp        |     return _bootstrap._gcd_import(name[level:], package, level)
webapp        |   File "<frozen importlib._bootstrap>", line 986, in _gcd_import
webapp        |   File "<frozen importlib._bootstrap>", line 969, in _find_and_load
webapp        |   File "<frozen importlib._bootstrap>", line 956, in _find_and_load_unlocked
webapp        | ImportError: No module named 'rest_framework_swagger'
``

pierotofy, 2016-12-05:
You have an old volume on your image, which contains an old node_modules/ folder; you need to remove that before you can rebuild the webapp image:

```
docker-compose down
rm -r node_modules/
docker-compose build --no-cache webapp
docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml up
```

macsee, 2016-12-05:
That did the trick! Thank you!
Cheers.

closed 2016-12-05 by pierotofy