WebODM/WebODM #62: postgresql-9.5-postgis-2.3 installation failed

desireB, 2016-12-06:
Hi,

I upgraded my system to Ubuntu 16.10, and I can't get WebODM to run.
I'm getting this error : 
`
E: Version '2.3.0+dfsg-2.pgdg80+1' for 'postgresql-9.5-postgis-2.3' was not found
E: Version '2.3.0+dfsg-2.pgdg80+1' for 'postgresql-9.5-postgis-2.3-scripts' was not found
E: Version '2.3.0+dfsg-2.pgdg80+1' for 'postgis' was not found
ERROR: Service 'db' failed to build: The command '/bin/sh -c apt-get update       && apt-get install -y --no-install-recommends            postgresql-$PG_MAJOR-postgis-$POSTGIS_MAJOR=$POSTGIS_VERSION            postgresql-$PG_MAJOR-postgis-$POSTGIS_MAJOR-scripts=$POSTGIS_VERSION            postgis=$POSTGIS_VERSION       && rm -rf /var/lib/apt/lists/*' returned a non-zero code: 100
`

Should I change something in the db/Dockerfile ?

Thank you.

pierotofy, 2016-12-06:
This seems like a problem with Docker itself rather than WebODM; have you searched on Google for how to properly update Docker after a dist-upgrade on Ubuntu?

pierotofy, 2016-12-06:
This might actually be a problem related to a package that doesn't exist anymore; the automated build just failed also: https://travis-ci.org/OpenDroneMap/WebODM

desireB, 2016-12-06:
I did a clean install of Ubuntu 16.10.
The issue was reported here also : https://github.com/appropriate/docker-postgis/issues/47 

Using 2.3.1+dfsg-1.pgdg80+1 might work. But it's using Postgres 9.6 .

pierotofy, 2016-12-06:
I've just updated the Dockerfile for the db image. Should work now.

```
git pull origin master
docker-compose build db
docker-compose -f ......
```

Thanks for reporting the issue!

closed 2016-12-06 by pierotofy

desireB, 2016-12-06:
Thank you for the prompt reaction and fix !

I was able to build db.

But I ran into 
`
fatal: destination path 'nodeodm/external/node-OpenDroneMap' already exists and is not an empty directory.
Clone of 'https://github.com/pierotofy/node-OpenDroneMap' into submodule path 'nodeodm/external/node-OpenDroneMap' failed
ERROR: Service 'webapp' failed to build: The command '/bin/sh -c git submodule update' returned a non-zero code: 1
`

I deleted all images, volumes and recreated db, and run the docker-compose -f ... , and I got the error.

Thank you !

pierotofy, 2016-12-06:
Similar issue to #61, make sure the external/node-OpenDroneMap directory is empty.

```
docker-compose down
rm -r nodeodm/external/node-OpenDroneMap/
docker-compose build --no-cache webapp
docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml up
```

desireB, 2016-12-06:
Thanks. That did fix it.
I'm having issues connecting to db service on 5432.
Getting : 
`
webapp        |     connection = Database.connect(**conn_params)
webapp        |   File "/usr/local/lib/python3.5/site-packages/psycopg2/__init__.py", line 164, in connect
webapp        |     conn = _connect(dsn, connection_factory=connection_factory, async=async)
webapp        | django.db.utils.OperationalError: could not connect to server: Connection refused
webapp        | 	Is the server running on host "localhost" (::1) and accepting
webapp        | 	TCP/IP connections on port 5432?
webapp        | could not connect to server: Connection refused
webapp        | 	Is the server running on host "localhost" (127.0.0.1) and accepting
webapp        | 	TCP/IP connections on port 5432?

`

ufw is disabled on my machine and I do have dnsmasq running.

Did anyone else run into this issue before ?

Thank you.

pierotofy, 2016-12-06:
Did you change webodm\settings.py or webodm\local_settings.py? Your DATABASES HOST should be 'db', not 'localhost'.

desireB, 2016-12-07:
Thanks ! Yes, at first, I wanted to run natively, and I created the webodm\local_settings.py file. I forgot about it. I remove it and it's working fine now. Thank you so much for your help.