WebODM/WebODM #65: Error messages on install

leszekmp, 2016-12-12:
Installed the Docker image on Windows 10 64-bit. In the terminal window are repeated error messages:

[##mwebapp exit with code 127  (where ## is an integer in the range 31 through 36)
No such file or directory[0m /usr/ bin/env: bash
No such file or directory/usr/ bin/env: bash
No such file or directory/usr/ bin/env: bash
(repeated a total of 10 times)

Looking at the webapp container log in Kitematic, the "No such file or directory/usr/ bin/env: bash" also shows up 11 times.

Took a while, but figured out how to open up the WebApp. Am processing a simple task, and it seems to be working. So are those error messages of concern?

pierotofy, 2016-12-12:
Perhaps :)

I'm guessing this is the `wait-for-it.sh` script. Would you be able to change this line from:

```
#!/usr/bin/env bash
#   Use this script to test if a given TCP host/port are available
```
To:

```
#!/bin/bash
#   Use this script to test if a given TCP host/port are available
```

And report whether the problem still happens?

leszekmp, 2016-12-12:
Thanks for the fast response. Not that up on Linux, but I'll try to figure out how to do that.

pierotofy, 2016-12-12:
You can do that from Windows, simply open the `wait-for-it.sh` script with Notepad (it's in the WebODM root directory), do the changes and rebuild the docker image using Kinematic.

leszekmp, 2016-12-12:
Took a while, but I figured out how to do that. Nope - I now get the following error message in the logs for the web app:

/bin/bash: /webodm/wait-for-it.sh: /bin/bash^M: bad interpreter: No such file or directory

in addition to the previous error message.

I am changing the script back to its previous form.

pierotofy, 2016-12-12:
Thank you for trying!

Would you be able to copy/paste the whole output of the terminal window so that I can try to identify the place where this error is happening?

leszekmp, 2016-12-12:
This is not exactly intuitive for a Windows guy :). Here's the output from the Docker terminal window, after I deleted the original Containers.

$ docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml up
Creating db
Creating node-odm-1
Creating webapp
Attaching to db, node-odm-1, webapp
[36mdb            |[0m The files belonging to this database system will be owned by user "postgres".
[36mdb            |[0m This user must also own the server process.
[36mdb            |[0m
[33mnode-odm-1    |[0m info: No tasks dump found
[36mdb            |[0m The database cluster will be initialized with locale "en_US.utf8".
[33mnode-odm-1    |[0m info: Checking for orphaned directories to be removed...
[36mdb            |[0m The default database encoding has accordingly been set to "UTF8".
[36mdb            |[0m The default text search configuration will be set to "english".
: No such file or directorysr/bin/env: bash
[36mdb            |[0m
[36mdb            |[0m Data page checksums are disabled.
[36mdb            |[0m
[36mdb            |[0m fixing permissions on existing directory /var/lib/postgresql/data ... ok
[36mdb            |[0m creating subdirectories ... ok
[36mdb            |[0m selecting default max_connections ... 100
[36mdb            |[0m selecting default shared_buffers ... 128MB
[36mdb            |[0m selecting dynamic shared memory implementation ... posix
[36mdb            |[0m creating configuration files ... ok
[36mdb            |[0m creating template1 database in /var/lib/postgresql/data/base/1 ... ok
[36mdb            |[0m initializing pg_authid ... ok
[36mdb            |[0m initializing dependencies ... ok
[36mdb            |[0m creating system views ... ok
[36mdb            |[0m loading system objects' descriptions ... ok
[36mdb            |[0m creating collations ... ok
[36mdb            |[0m creating conversions ... ok
[36mdb            |[0m creating dictionaries ... ok
[36mdb            |[0m setting privileges on built-in objects ... ok
[36mdb            |[0m creating information schema ... ok
[36mdb            |[0m loading PL/pgSQL server-side language ... ok
[36mdb            |[0m vacuuming database template1 ... ok
[36mdb            |[0m copying template1 to template0 ... ok
[36mdb            |[0m copying template1 to postgres ... ok
[36mdb            |[0m syncing data to disk ...
[36mdb            |[0m WARNING: enabling "trust" authentication for local connections
[36mdb            |[0m You can change this by editing pg_hba.conf or using the option -A, or
[36mdb            |[0m --auth-local and --auth-host, the next time you run initdb.
[36mdb            |[0m ****************************************************
[36mdb            |[0m WARNING: No password has been set for the database.
[36mdb            |[0m          This will allow anyone with access to the
[36mdb            |[0m          Postgres port to access your database. In
[36mdb            |[0m          Docker's default configuration, this is
[36mdb            |[0m          effectively any other container on the same
[36mdb            |[0m          system.
[36mdb            |[0m
[36mdb            |[0m          Use "-e POSTGRES_PASSWORD=password" to set
[36mdb            |[0m          it in "docker run".
[36mdb            |[0m ****************************************************
[36mdb            |[0m ok
[36mdb            |[0m
[36mdb            |[0m Success. You can now start the database server using:
[36mdb            |[0m
[36mdb            |[0m     pg_ctl -D /var/lib/postgresql/data -l logfile start
[36mdb            |[0m
[36mdb            |[0m waiting for server to start....LOG:  database system was shut down at 2016-12-12 22:11:17 UTC
[36mdb            |[0m LOG:  MultiXact member wraparound protections are now enabled
[36mdb            |[0m LOG:  database system is ready to accept connections
[36mdb            |[0m LOG:  autovacuum launcher started
[36mdb            |[0m  done
[36mdb            |[0m server started
[36mdb            |[0m ALTER ROLE
[36mdb            |[0m
[36mdb            |[0m
[36mdb            |[0m /docker-entrypoint.sh: running /docker-entrypoint-initdb.d/init-db.sql
[... 23 lines cut ...]
: No such file or directorym /usr/bin/env: bash
: No such file or directorysr/bin/env: bash
: No such file or directorysr/bin/env: bash
: No such file or directorysr/bin/env: bash
: No such file or directorysr/bin/env: bash
[31mwebapp exited with code 127
: No such file or directorym /usr/bin/env: bash
: No such file or directorysr/bin/env: bash
: No such file or directorysr/bin/env: bash
: No such file or directorysr/bin/env: bash
: No such file or directorysr/bin/env: bash
: No such file or directorysr/bin/env: bash
: No such file or directorysr/bin/env: bash
[34mwebapp exited with code 127
: No such file or directory[0m /usr/bin/env: bash
: No such file or directory/usr/bin/env: bash
: No such file or directory/usr/bin/env: bash
: No such file or directory/usr/bin/env: bash
: No such file or directory/usr/bin/env: bash
: No such file or directory/usr/bin/env: bash
: No such file or directory/usr/bin/env: bash
: No such file or directory/usr/bin/env: bash
: No such file or directory/usr/bin/env: bash
[36;1mwebapp exited with code 127
[0m

pierotofy, 2016-12-14:
@leszekmp would you be able to download the latest version of WebODM (`git pull origin master`) and rebuild your docker images? Does the error message still appear?

leszekmp, 2016-12-14:
As a newbie to these apps, I'm not exactly sure what I should do. Do I need to remove the containers from Kitematic? Tried running the command you gave in the Docker Quickstart Terminal (I'm running Docker Toolbox, and Kitematic invokes VirtualBox, which I assume is because my system doesn't have Hyper-V support), and I don't think I'm using it right.   I'm assuming you need to specify WebODM somehow, but it's not included in the command, and just typing it in as above gives the error message "fatal: Not a git repository (or any of the parent directories): .git".

pierotofy, 2016-12-14:
Yes, remove the containers from kinematics.

From the Docker Quickstart Terminal, cd to the location of WebODM.

```bash
# cd path/to/WebODM
# git pull origin master
```

Or without using the command line, open Notepad, create a file with the following:

```
git pull origin master
pause
```

And save it as "git-pull.bat" in your WebODM directory. Then double-click "git-pull.bat". That will update your version of WebODM. Then rebuild the containers with Kinematic.

Let me know if you need further instructions!

leszekmp, 2016-12-14:
Got a message saying that I couldn't do that, since I had modified wait-for-it.sh and hadn't merged/stashed it. Figured that out, followed your directions, and now the "bash" error message is gone.

leszekmp, 2016-12-14:
Thanks for your help. I've been working with PCs for 35+ years, and you can never get caught up with everything out there.

pierotofy, 2016-12-14:
Yay, glad it's fixed. Thanks again for reporting the problem.

And don't worry, I think everyone in this field (myself included) struggles to keep up with all the latest changes. :)

closed 2016-12-14 by pierotofy