WebODM/WebODM #70: Proccesing error

esp4ik, 2016-12-19:
Hi everyone!

I tried to run task from dashboard and get an error. Does anybody know how to fix that error?
P.S. I have installed WebODM with Docker.

Traceback (most recent call last):
  File "/code/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
    command.run(args)
  File "/code/SuperBuild/src/opensfm/opensfm/commands/mesh.py", line 22, in run
    reconstructions = data.load_reconstruction()
  File "/code/SuperBuild/src/opensfm/opensfm/dataset.py", line 336, in load_reconstruction
    with open(self.__reconstruction_file(filename)) as fin:
IOError: [Errno 2] No such file or directory: '/var/www/data/29ae05a3-3fe6-4be2-9e10-45a98e67e331/opensfm/reconstruction.json'
[ERROR]   quitting cause: 
	PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/run_all /var/www/data/29ae05a3-3fe6-4be2-9e10-45a98e67e331/opensfm
returned with code 256.

pierotofy, 2016-12-19:
Hi!

Would you be able to share your images dataset? It might be that you don't have enough overlap, or not enough recognizable features.

esp4ik, 2016-12-19:
Here is images
1) https://www.dropbox.com/s/2khbcr9ulh15htm/1.zip?dl=0
2) https://www.dropbox.com/s/hk7zxgz3adzrucm/2.zip?dl=0

pierotofy, 2016-12-19:
Yeah because of the low number of images the reconstruction is failing, probably because of too few matches.

@dakotabenjamin this seems to be a common occurrence (5 images fail with the above error message). Is there a command line parameter that needs to be tweaked when passing a small number of images that can lead to a proper reconstruction?

esp4ik, 2016-12-19:
Thanks for help! It works!

pierotofy, 2016-12-20:
@esp4ik would you mind sharing how you made it work? Did you just use more images?

esp4ik, 2016-12-20:
Yes, I just added more images - 14.

closed 2016-12-20 by esp4ik

dakotabenjamin, 2016-12-20:
I am not involved in the development process over at mapillary/opensfm so I can't say whether very small datasets has been a use case for them.