WebODM/WebODM #71: "All processing nodes are offline" message

pierotofy, 2016-12-19:
If you keep the dashboard open for longer than 5 minutes, any upload will result in a "all processing nodes are offline" message. This is because the processing nodes information is not refetched after the first load of the app. This needs to be fixed.

BKLronin, 2016-12-22:
Got something similar "There are no processing nodes available. Make sure at least one of them is reachable." with this in the CLI

[0m Starting development server at http://0.0.0.0:8000/
[32mwebapp        |[0m Quit the server with CONTROL-C.
[32mwebapp        |[0m INFO Tasks pending, set project 1 deleting flag
[32mwebapp        |[0m WARNING Not Found: /api/projects/
[32mwebapp        |[0m WARNING "GET /api/projects/?ordering=-created_at&page=0 HTTP/1.1" 404 26
[32mwebapp        |[0m INFO Acquiring lock: Task ID: 1
[32mwebapp        |[0m INFO Removing task Task ID: 1
[32mwebapp        |[0m WARNING Not Found: /api/projects/
[32mwebapp        |[0m WARNING "GET /api/projects/?ordering=-created_at&page=0 HTTP/1.1" 404 26
[32mwebapp        |[0m INFO Deleted 1 projects
[32mwebapp        |[0m INFO Acquiring lock: Task ID: 4
[32mwebapp        |[0m INFO Removing task Task ID: 4
[32mwebapp        |[0m WARNING Not Found: /api/projects/4/tasks/4/
[32mwebapp        |[0m WARNING "GET /api/projects/4/tasks/4/ HTTP/1.1" 404 23
[32mwebapp        |[0m INFO Deleted project 4
[32mwebapp        |[0m INFO Tasks pending, set project 3 deleting flag
[32mwebapp        |[0m INFO Tasks pending, set project 2 deleting flag
[32mwebapp        |[0m WARNING Not Found: /api/projects/
[32mwebapp        |[0m WARNING "GET /api/projects/?ordering=-created_at&page=0 HTTP/1.1" 404 26

pierotofy, 2016-12-22:
If you refresh the dashboard page does it work? Just trying to confirm this is the same problem.

BKLronin, 2016-12-22:
It seems to be stuck somehow. I restarted the browser and webodm itself but it won´t let me use the standard node. I can verify it is available though the webinterface and the cli.

pierotofy, 2017-01-21:
It looks like it might be just a timing issue related to when docker decides to spin up the node-odm-1 processing node. If node-odm-1 comes online AFTER webodm_app has started, webodm_app will not get a chance to refresh the "last seen online" field until the next scheduled update (which happens every 30 seconds), so if in the meanwhile a user spins up the app, opens the dashboards and uploads some files (all within 30 seconds), you'll get the error. 

My proposed fix includes specifying that the webodm_app image depends on node-odm-1, and node-odm-1 should always start before webodm_app (right now it's a race). I can rule out a javascript problem, since loadProcessingNodes() is actually called everytime a new upload happens, so it should be in sync with the database.

I will close this for now. If people continue having the issue even after this fix, we'll reopen it.

commit 33ba5312 (2017-01-24): Backend layout fix, specified node-odm-1 as a dependency of webapp to fix #71
closed 2017-01-21 by pierotofy