WebODM/WebODM #72: Install issues and halt after system check

BKLronin, 2016-12-20:
Hi,
the Installation was a bit choppy for me. It took a few moments to realize that docker runs a vm. Than I was a Little confused why python and all that Needs to be installed outside of that vm. I thought that docker runs an Image with all necessaritys. I was wrong.

I managed to get to the System check but only with the direct download of the zip file. It gave me error Code 126 when downloading via github. 

I can´t get pass the System check and I don´t knowo if it is already running and if yes at which adress? 

Those are the last lines before nothing happens:

[32mwebapp        |[0m INFO Starting background scheduler...
[32mwebapp        |[0m Operations to perform:
[32mwebapp        |[0m   Apply all migrations: admin, app, auth, contenttypes, guardian, nodeodm, sessions
[32mwebapp        |[0m Running migrations:
[32mwebapp        |[0m   No migrations to apply.
[32mwebapp        |[0m Python 3.5.2 (default, Nov 17 2016, 22:42:56)
[32mwebapp        |[0m [GCC 4.9.2] on linux
[32mwebapp        |[0m Type "help", "copyright", "credits" or "license" for more information.
[32mwebapp        |[0m (InteractiveConsole)
[32mwebapp        |[0m >>> (<ProcessingNode: node-odm-1:3000>, False)
[32mwebapp        |[0m >>>
[32mwebapp        |[0m Performing system checks...
[32mwebapp        |[0m
[32mwebapp        |[0m INFO Starting background scheduler...
[32mwebapp        |[0m System check identified no issues (0 silenced).

pierotofy, 2016-12-20:
Looks like WebODM started properly. Find the IP of your docker-engine machine by running:

```
docker-machine ip
```

Then open a browser to http://<docker-machine-ip>:8000

Does it work?

Python needs to be installed because docker-compose is a Python script. Git needs to be installed so that people can download the code from Github. If it wasn't for docker you would have to install a lot more dependencies on top of those :)

BKLronin, 2016-12-20:
It works, I logged in and will test right now. :)
All good, thanks for your help.

pierotofy, 2016-12-20:
Thanks for reporting the issue!

closed 2016-12-20 by pierotofy