WebODM/WebODM #74: Processing quits when no GPS in exif

BKLronin, 2016-12-22:
I tried to stich a sample dataset with no GPS coordinates in the exif. I have seen the appropiate entry in the JSON file configuring the node but I can´t edit it or find it within the folders. 

[DEBUG]   running /code/build/bin/odm_extract_utm -imagesPath /var/www/data/a9f0cfb9-ee28-40ae-a256-d1f9936bf71e/images_resize/ -imageListFile /var/www/data/a9f0cfb9-ee28-40ae-a256-d1f9936bf71e/opensfm/list_r000.out -outputCoordFile /var/www/data/a9f0cfb9-ee28-40ae-a256-d1f9936bf71e/odm_georeferencing/coords.txt -logFile /var/www/data/a9f0cfb9-ee28-40ae-a256-d1f9936bf71e/odm_georeferencing/odm_georeferencing_utm_log.txt
Error in OdmExtractUtm:
No valid GPS position found
For more detailed information, see log file.
[ERROR]   quitting cause: 
	/code/build/bin/odm_extract_utm -imagesPath /var/www/data/a9f0cfb9-ee28-40ae-a256-d1f9936bf71e/images_resize/ -imageListFile /var/www/data/a9f0cfb9-ee28-40ae-a256-d1f9936bf71e/opensfm/list_r000.out -outputCoordFile /var/www/data/a9f0cfb9-ee28-40ae-a256-d1f9936bf71e/odm_georeferencing/coords.txt -logFile /var/www/data/a9f0cfb9-ee28-40ae-a256-d1f9936bf71e/odm_georeferencing/odm_georeferencing_utm_log.txt
returned with code 256.

An error occurred. Check stdout above or the logs.

dakotabenjamin, 2016-12-22:
There is currently a fix on the way at https://github.com/OpenDroneMap/OpenDroneMap/pull/440

BKLronin, 2016-12-22:
Excellent, can be helpful when capturing buildings from the ground with no GPS attached.

pierotofy, 2017-01-04:
Closing this since a fix will be available.

closed 2017-01-04 by pierotofy