WebODM/WebODM #77: Fails to run on Docker Cloud

dbrosy, 2017-01-14:
I have deployed via Docker Cloud but the container stops after starting.
https://cloud.docker.com

Received event 'die' from node ...

i have my own server in docker cloud but cannot run it, is this possible. I have a separate server with docker and when i run "webodm.sh start" it works as expected

pierotofy, 2017-01-15:
It's difficult to say what is causing the problem, but I'm guessing this is probably something specific to docker cloud rather than an issue with WebODM itself.

Perhaps if you have a more detailed error log I could try to help. I have never used docker cloud.

dbrosy, 2017-01-16:
I am trying to investigate the issue but I also have limited Docker / Docker cloud experience so it may take some time.

pierotofy, 2017-02-20:
I'll close this for now.

closed 2017-02-20 by pierotofy