WebODM/WebODM #81: WebODM failed to start

desireB, 2017-01-24:
Hi,

Thanks for the exciting new features !

I'm getting this error after running update, and start :+1: 
`
bapp        | 
webapp        | INFO Created default group
webapp        | Unhandled exception in thread started by <function check_errors.<locals>.wrapper at 0x7f6f510318c8>
webapp        | Traceback (most recent call last):
webapp        |   File "/usr/local/lib/python3.5/site-packages/django/utils/autoreload.py", line 226, in wrapper
webapp        |     fn(*args, **kwargs)
webapp        |   File "/usr/local/lib/python3.5/site-packages/django/core/management/commands/runserver.py", line 121, in inner_run
webapp        |     self.check(display_num_errors=True)
webapp        |   File "/usr/local/lib/python3.5/site-packages/django/core/management/base.py", line 385, in check
webapp        |     include_deployment_checks=include_deployment_checks,
webapp        |   File "/usr/local/lib/python3.5/site-packages/django/core/management/base.py", line 372, in _run_checks
webapp        |     return checks.run_checks(**kwargs)
webapp        |   File "/usr/local/lib/python3.5/site-packages/django/core/checks/registry.py", line 81, in run_checks
webapp        |     new_errors = check(app_configs=app_configs)
webapp        |   File "/usr/local/lib/python3.5/site-packages/django/core/checks/urls.py", line 14, in check_url_config
webapp        |     return check_resolver(resolver)
webapp        |   File "/usr/local/lib/python3.5/site-packages/django/core/checks/urls.py", line 24, in check_resolver
webapp        |     for pattern in resolver.url_patterns:
webapp        |   File "/usr/local/lib/python3.5/site-packages/django/utils/functional.py", line 35, in __get__
webapp        |     res = instance.__dict__[self.name] = self.func(instance)
webapp        |   File "/usr/local/lib/python3.5/site-packages/django/urls/resolvers.py", line 310, in url_patterns
webapp        |     patterns = getattr(self.urlconf_module, "urlpatterns", self.urlconf_module)
webapp        |   File "/usr/local/lib/python3.5/site-packages/django/utils/functional.py", line 35, in __get__
webapp        |     res = instance.__dict__[self.name] = self.func(instance)
webapp        |   File "/usr/local/lib/python3.5/site-packages/django/urls/resolvers.py", line 303, in urlconf_module
webapp        |     return import_module(self.urlconf_name)
webapp        |   File "/usr/local/lib/python3.5/importlib/__init__.py", line 126, in import_module
webapp        |     return _bootstrap._gcd_import(name[level:], package, level)
webapp        |   File "<frozen importlib._bootstrap>", line 986, in _gcd_import
webapp        |   File "<frozen importlib._bootstrap>", line 969, in _find_and_load
webapp        |   File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
webapp        |   File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
webapp        |   File "<frozen importlib._bootstrap_external>", line 665, in exec_module
webapp        |   File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
webapp        |   File "/webodm/webodm/urls.py", line 23, in <module>
webapp        |     url(r'^', include('app.urls')),
webapp        |   File "/usr/local/lib/python3.5/site-packages/django/conf/urls/__init__.py", line 50, in include
webapp        |     urlconf_module = import_module(urlconf_module)
webapp        |   File "/usr/local/lib/python3.5/importlib/__init__.py", line 126, in import_module
webapp        |     return _bootstrap._gcd_import(name[level:], package, level)
webapp        |   File "<frozen importlib._bootstrap>", line 986, in _gcd_import
webapp        |   File "<frozen importlib._bootstrap>", line 969, in _find_and_load
webapp        |   File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
webapp        |   File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
webapp        |   File "<frozen importlib._bootstrap_external>", line 665, in exec_module
webapp        |   File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
webapp        |   File "/webodm/app/urls.py", line 23, in <module>
webapp        |     boot()
webapp        |   File "/webodm/app/boot.py", line 26, in boot
webapp        |     default_group.permissions.add(Permission.objects.get(codename="view_processingnode"))
webapp        |   File "/usr/local/lib/python3.5/site-packages/django/db/models/manager.py", line 85, in manager_method
webapp        |     return getattr(self.get_queryset(), name)(*args, **kwargs)
webapp        |   File "/usr/local/lib/python3.5/site-packages/django/db/models/query.py", line 385, in get
webapp        |     self.model._meta.object_name
webapp        | django.contrib.auth.models.DoesNotExist: Permission matching query does not exist.

`

pierotofy, 2017-01-24:
Thanks for reporting the problem! I think @dakotabenjamin had the same problem yesterday.

Remove your WebODM directory and start the `./webodm.sh rebuild && ./webodm.sh start` from a fresh clone (`git clone https://github.com/OpenDroneMap/WebODM/`). You likely have some old migrations lingering around.

desireB, 2017-01-24:
Thanks ! That did solve it ;) Running on a dataset now.

closed 2017-01-24 by desireB