WebODM/WebODM #82: WebODM Python Error

AnthonyCoughlin, 2017-01-27:
All,

If there is a better place to post an issue like this please advise.  See the attached image.  I am having an issue starting WebODM via Docker.  When running the initial checks it fails to find Python.  In an effort to find the resolution myself I've ensured that I have Python 3.x installed.  Restarted the machine and and attempted to run the ./webodm.sh start command again with the same error.  Please let me know if you need any additional information.

Thanks in advance,
Anthony

[image]

pierotofy, 2017-01-27:
Hi @AnthonyCoughlin thanks for reporting the issue.

If you run the following command:
```
python --version
```
What is the output?

If you get a "command not found" error, make sure that you have added your python executable paths (c:\python3x\bin and c:\python3x\scripts ?) to your PATH environment variable. http://www.computerhope.com/issues/ch000549.htm

Then restart the Docker Quickstart Terminal and try again.

AnthonyCoughlin, 2017-01-28:
pierotofy

Thank you Sir.  I actually ran a repair install on python 3.x and found the 'set environment variables' was unchecked.  Checked it, completed install, reopened Docker and the './webodm.sh start' ran.  Unfortunately I'm receiving one last error preventing WebODM from starting.  I'll make a new post to address this error

Thanks again for your help,
Anthony

closed 2017-01-28 by AnthonyCoughlin