WebODM/WebODM #83: ERROR: Validation failed, reason(s): webapp.depends_on value ['db', 'db', 'node-odm-1'] has non-unique elements

McStrav, 2017-01-27:
Hi,

Had no problems with previous build, i.e. prior to Potree implementation.  

I thought I would give this latest build a try and I'm getting a validation error when running 

`./webodm.sh start`

output as follows:

> Checking for docker...   OK
> Checking for git...   OK
> Checking for python...   OK
> Checking for pip...   OK
> Checking for docker-compose...   OK
> Starting WebODM...
> docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml up
> ERROR: Validation failed, reason(s):
> webapp.depends_on value ['db', 'db', 'node-odm-1'] has non-unique elements

Any ideas?

Thanks,

Eamonn

pierotofy, 2017-01-27:
Hi @McStrav, thanks for reporting the problem! I have made some changes to the docker-compose.nodeodm.yml file just now, could you please try to run:

```
git pull origin master
./webodm.sh start
```

And see if the problem is fixed?

Thanks!

McStrav, 2017-01-28:
Hi @pierotofy that seems to have fixed it... 

I killed and removed all docker containers and images, then re-ran:

```
git clone https://github.com/OpenDroneMap/WebODM
cd WebODM
./webodm.sh start
```

It's now all up and running.

Thanks...

closed 2017-01-28 by McStrav