WebODM/WebODM #84: Cannot create container for service webapp:  invalid bind mount spec

AnthonyCoughlin, 2017-01-28:
All,

See attached screenshot.  Receiving an error after running the ./webodm.sh start' command.  Cannot create container for service webapp: invalid bind mount spec'.  There are additional warnings a few lines up as well.  Thanks for all the help thus far!

Anthony

[image]

pierotofy, 2017-01-28:
@AnthonyCoughlin, could you try to run:

```
MSYS_NO_PATHCONV=1 ./webodm.sh start
```

?

pierotofy, 2017-01-28:
Alternatively, also try:

```
COMPOSE_CONVERT_WINDOWS_PATHS=1 ./webodm.sh start
```

Do any of these two solve the problem?

AnthonyCoughlin, 2017-01-28:
pierotofy,

See attached screenshot.  Thanks again, I was reading about the windows/msys path errors and conversion after searching the error.  

It seems as though it worked but now its freezing when trying to access /webodm/wait-for-it.sh.  I've been researching this for a bit but could use some assistance. 

 I'm not going to try to pretend here....  I'm super new to this!  So please understand how appreciative I am for all the help.

[image]

AnthonyCoughlin, 2017-01-28:
Searching the error (bad interpreter:) - I've been seeing a dos2unix dependency that can be installed on docker.  Do you recommend this approach?

pierotofy, 2017-01-28:
No worries :) I think we are all learning how to troubleshoot these errors at the moment.

Just to rule something out, did you modify wait-for-it.sh while troubleshooting previous problems? If so, try to revert it back by issuing `git checkout wait-for-it.sh`. If you have modified it on Windows, you probably have inadvertently changed the line endings, which are going to confuse bash.

pierotofy, 2017-01-28:
Another cause could be that your git settings converted the line endings. Try to clone the repo with:

`git clone http://github.com/OpenDroneMap/WebODM --config core.autocrlf=input`

I think `dos2unix` could work, but it's one more dependency that I'd rather not install as part of the process. Let's explore some other options first.

AnthonyCoughlin, 2017-01-28:
Ok, no dos2unix, got it.

I didn't modify anything that I know of.  My workflow after the initial error: connect to the WebODM folder and run the ./webodm.sh with the prefix you provided (COMPOSE_CONVERT_WINDOWS_PATHS=1 ./webodm.sh start).  That's when I received the 'bad interpreter' error.

***(For the assistance of others your first recommendation line of code didn't work for me - MSYS_NO_PATHCONV=1 ./webodm.sh start)***

I'll test the clone recommendation and let you know what happens!  Thanks again!

AnthonyCoughlin, 2017-01-28:
pierotofy,

See attached.  Success!  Thanks again for your assistance.  Couldn't have done it with out you!

Workflow:
I deleted the existing \webodm directory using `rm -rf \webodm`
Re-cloned with `git clone http://github.com/OpenDroneMap/WebODM --config core.autocrlf=input`
Navigated to the webodm folder `cd \webodm`
Ran `COMPOSE_CONVERT_WINDOWS_PATHS=1 ./webodm.sh start`
WebODM started

CANNOT WAIT TO START PLAYING WITH THIS APP!!

Feel free to close this one out.

Anthony

pierotofy, 2017-01-29:
@AnthonyCoughlin I'm glad it worked!

I have modified the Readme and added the COMPOSE_CONVERT_WINDOWS_PATHS to the webodm.sh script, so hopefully this will improve the install process.

Thanks for reporting the problem!

closed 2017-01-29 by pierotofy