WebODM/WebODM #86: WebODM hanging up at chmod

AnthonyCoughlin, 2017-02-04:
Good day folks,

See attached.  I've cloned the most recent version of WebODM (today), and when executing the `./webodm.sh start` WebODM hangs up when trying to access chmod.  Says the file does not exist.

@pierotofy, I'm afraid this might be a pathway issue again.  I've tried running each of the following when attempting to start with no luck:

`./webodm.sh start`
`MSYS_NO_PATHCONV=1 ./webodm.sh start`
`COMPOSE_CONVERT_WINDOWS_PATHS=1 ./webodm.sh start`

Thanks again for the help!

[image]

[image]

pierotofy, 2017-02-04:
@AnthonyCoughlin thanks for reporting the issue.

Would you be able to provide additional details about how you cloned the repository? Did you use:

`git clone https://github.com/OpenDroneMap/WebODM --config core.autocrlf=input`

?

Also note that after an update of the WebODM code, you'll need to run:

```
./webodm.sh update
```

That command will take care of rebuilding all docker images related to WebODM.

AnthonyCoughlin, 2017-02-04:
@pierotofy, yes, I ran using `--config core.autocrl=input`.  Didn't realize I needed to run `./webodm.sh` when cloning new folders.  Got it running, thanks again.

Side note:  See attached below.  I received this warning wasn't sure if it was of any importance:

Thanks again for your help,
Anthony

[image]

pierotofy, 2017-02-04:
Glad it worked!

The warning is not important, it's an optional dependency, thanks for pointing it out though!

closed 2017-02-04 by pierotofy