WebODM/WebODM #87: CCD width missing

AnthonyCoughlin, 2017-02-04:
@pierotofy 

I'm flying the Autel X-Star Premium.  When I process photos through WebODM, I noticed that the CCD width is missing.  How do I retrieve that info to force WebODM to use it or add it to the WebODM list.

Thanks again,
Anthony

[image]

pierotofy, 2017-02-04:
You can use the "force-ccd" option from the "Advanced" tab when you upload images to start a new task.

[image]

[image]

See this link for more information about calculating the CCD width. http://phototour.cs.washington.edu/focal.html

AnthonyCoughlin, 2017-02-05:
Thanks again.

closed 2017-02-05 by pierotofy

AnthonyCoughlin, 2017-02-05:
I believe I've got this one figured out.  Thanks.