WebODM/WebODM #90: Ability to download data even on failure

dakotabenjamin, 2017-02-07:
I would like to be able to download the point cloud or textured model even if the orthophoto fails, or anything really. Can we expose the "download all data" link even when it fails?

pierotofy, 2017-02-07:
It could be possible to add such functionality, but the changes would have to start to be added on node-OpenDroneMap first. Right now if any part of the process fails, node-OpenDroneMap will report a status of failed.

smathermather, 2017-02-07:
So changes to node-OpenDroneMap that give different status flags for different types of failure?

pierotofy, 2017-02-07:
Mm, I don't think that will be necessary, WebODM can simply download all assets and check for itself what has been created and what not. We just need to change start() https://github.com/pierotofy/node-OpenDroneMap/blob/master/libs/Task.js#L219 so that even if an orthophoto or a point cloud is not generated, that an output is still created (and skip tiling or potreeconverter when necessary). Right now it's going to fail if any of the assets is missing.

pierotofy, 2017-03-09:
Will probably tackle this next week.

pierotofy, 2017-03-14:
@dakotabenjamin @smathermather do you have an example dataset that fails to generate an orthophoto for testing purposes? Or is it just any dataset that lacks GPS/GPC info?

pierotofy, 2017-03-14:
Another question, I understand that sometimes the orthophoto might not be generated, but a point cloud does. If there's no point cloud, then there's also no orthophoto. What would be the use case of allowing a user to download results if no point cloud and no orthophoto is generated?

smathermather, 2017-03-14:
Anything without GPS ephemeris will do.

I can't think of a use case where you'd want the data if there were no point cloud.

pierotofy, 2017-03-16:
Fixed in #118

PR #118 (2017-03-16): Test task processing without orthophoto results, addition of availabl…
Fixes #90 

@dakotabenjamin would you have some time to look over the changes sometimes :smile: ?
closed 2017-03-16 by pierotofy