WebODM/WebODM #92: Entwine + Grayhound Plugin

pierotofy, 2017-02-08:
It would be cool to add the ability to link an external entiwine/grayhound server to WebODM so that point cloud results can be automatically added and merged into the index.

pierotofy, 2017-03-09:
I wonder if an efficient system would be to simply allow users to configure web hooks that would be triggered on certain events (on task completion, on task deletion, on task restart, etc.).

Once you have that, you don't need pooling and a server running entwine / grayhound or openarialmap could simply retrieve/index the assets it needs via API calls.

smathermather, 2017-03-09:
On a related note, I've been thinking about entwine / greyhound the last few days. FWIW, we are not far away from having the capacity to process datasets large enough that without LOD, streaming, etc. the current viewer won't be able to display the data produced.

pierotofy, 2019-06-08:
Entwine is now integrated in ODM and WebODM, so this can be closed. Greyhound is no longer needed and datasets can be read with Potree.

closed 2019-06-08 by pierotofy