WebODM/WebODM #100: WebODM hanging up

AnthonyCoughlin, 2017-02-11:
While running a task on 38 images WebODM seem so be stuck in a loop.  Says its unable to close temp stats file.  The attached message from Docker repeated several times while trying to execute [ROOT_HAHOG] on one of the images in WebODM (see attached below).

[image]

I restarted Docker then my physical machine.  Restarted Docker and WebODM now receiving the following error:

[image]

As always, Thanks for the help.

pierotofy, 2017-02-11:
Looks like you are running out of space! Make sure you have enough disk space.

AnthonyCoughlin, 2017-02-11:
I don't believe this is the case (see screenshot below), or am I looking in the wrong place??

[image]

I just wiped the WebODM folder again, cloned and upon updating got this error (see attached below).
.  I'm going to delete the folder and restart my machine again.  I'm getting close to uninstalling Docker and reinstalling.  I'll keep you updated.

[image]

@pierotofy, can't thank you enough for the quick responses!

AnthonyCoughlin, 2017-02-11:
Uninstalled Docker, restarted machine, and reinstalled Docker.  Deleted entire existing WebODM dir and recloned.  Ran WebODM update.  Back up and running.  Testing the same data set.  I'll keep you up to date.  Thanks again for the help.

AnthonyCoughlin, 2017-02-11:
Everything seems to be running fine with regard to WebODM starting now.  However, I'm experiencing a new error: 'refinePatchPFGS failed'.  I'm going to start a new issue.

closed 2017-02-11 by AnthonyCoughlin