WebODM/WebODM #101: refineBatchPFGS failed

AnthonyCoughlin, 2017-02-11:
WebODM stuck in a loop.  Watching Docker the command line is still running but has been repeating the same 'warnings' and WebODM is throwing an error message 'refineBatchPFGS failed'.  See attached.

[image]

[image]

Let me know if you need more info!

AnthonyCoughlin, 2017-02-12:
Still going.

[image]

pierotofy, 2017-02-12:
It is possible that the ReadTimeout error stopped execution. I'll have to investigate to tell for sure. refinePatchesBFGS often times fails during processing, but should not interrupt the process.

Restarting the task might fix it temporarily.

AnthonyCoughlin, 2017-02-12:
@pierotofy  Thanks again for the assistance.  When this happened it corrupted the entire Docker db.  I've received db errors when trying to restart the Docker Machine.  I ended up wiping the WebODM folder again and uninstalling Docker Toolbox.  Then reinstalling Docker Toolbox and recloned WebODM.  I had to run `docker-compose build` to repair the db (not sure why I thought the uninstall would have removed all files) then ran `./webodm.sh update`.  Now WebODM starts with no errors.  However, I attempted to stitch the same dataset today (FYI - I uploaded the same data to MapsMadeEasy.com last night and it worked just fine - I'm pro ODM tho man!) and it failed.  I received this error (see below).  I am forcing focal-length and CCD width, I may be putting in the wrong values - would that cause an error?

[image]

AnthonyCoughlin, 2017-02-12:
Also, watching the Docker cmd line, I keep seeing this warning (see below).  No sure if this helps you at all...

[image]

pierotofy, 2017-02-12:
@AnthonyCoughlin thanks for the additional screens! No worries, I use mapsmadeeasy on occasion as well :)

It looks like the same problem is happening (ReadTimeout error). I will have time to look at this sometimes next week, I'll post an update with hopefully a fix then.

AnthonyCoughlin, 2017-02-13:
@pierotofy, take your time.  Thanks again.

AnthonyCoughlin, 2017-02-13:
@pierotofy,

Update:  I'm starting to believe there is something wrong with my dataset.  I successfully tested four other datasets yesterday evening, using both default settings and forced focal and CCD.  I'm planning on testing the failing dataset this evening and will get back with you.

pierotofy, 2017-02-15:
I think the root cause of this problem lies in node-OpenDroneMap, and likely not on WebODM. Even though you received some error messages, the execution should have continued at the next cycle since the errors were thrown from a separate thread. Would you be able to share the dataset that you used?

AnthonyCoughlin, 2017-02-16:
#104 

@pierotofy, Should I post or reference this issue/bug on the ODM git?  Whats the best way to share my images?

pierotofy, 2017-02-16:
@AnthonyCoughlin Dropbox, or any other file sharing website that you like. If you can share the dataset publicly, just post the link here on Github. Otherwise send it here: https://www.masseranolabs.com/contact/.

I'll open an issue in the proper repository once I can reproduce it and confirm it.

AnthonyCoughlin, 2017-02-18:
Hey @pierotofy,  Sorry for the delay. Sent you a link to images via the above mentioned link.  Thanks again for your help!

pierotofy, 2017-02-18:
Got it.

I just ran the dataset and it processed correctly using the latest WebODM/node-OpenDroneMap.

[image]

I did use the code that was introduced in #103, so perhaps that could have made a difference, but I don't think it did. My guess is that perhaps your Docker containers don't have enough memory allocated, which is making the processing really, really slow.

Power down the Virtualbox "boot2docker" or "default" virtual machine from the VirtualBox Manager application, and follow these instructions to add more RAM: http://superuser.com/questions/926339/how-to-change-the-ram-allocated-to-an-os-in-virtualbox

Also, what happens if you add this processing node and use that instead? hostname: nodeodm.masseranolabs.com port: 80?

AnthonyCoughlin, 2017-02-18:
I'll test this afternoon.  Thanks!

pierotofy, 2017-02-20:
I'll close this for now, unless the problem is still there, in which case we can reopen it.

closed 2017-02-20 by pierotofy