WebODM/WebODM #99: Error Invalid HTTP_HOST header:

AnthonyCoughlin, 2017-02-11:
I've recently cloned and updated webodm to run the newest version.  Now receiving this error.  Thanks for the help.

[image]

pierotofy, 2017-02-11:
Hi @AnthonyCoughlin thanks for reporting the problem!

This error was introduced from the Django core update. Nice catch!

I have made some changes to settings.py by adding a value of '*' in the ALLOWED_HOSTS variable. This should fix the error you are experiencing.

AnthonyCoughlin, 2017-02-11:
That's what I read earlier, something like `ALLOWED_HOSTS = ['*']` should allow all hosts.  Thanks for the quick response @pierotofy.

Running `./webodm.sh update` and will attempt to restart WebODM, then let you know the result.

AnthonyCoughlin, 2017-02-11:
Up and running!  Thanks again for the fast response @pierotofy

closed 2017-02-11 by AnthonyCoughlin