WebODM/WebODM #102: Python Bindings

pierotofy, 2017-02-14:
It would facilitate adoption if there were Python bindings to access the API.

rmallermartins, 2017-08-18:
I'm thinking about creating a Python Client (for accessing and using the api) for WebODM. And even a Django Package after that, with Login and Host configuration in the settings.

Help would be appreciated, if anyone is interested :)

pierotofy, 2017-08-18:
This would be awesome. I'd be happy to help (answering questions about the API, reviewing code, testing, providing feedback).

rmallermartins, 2017-08-18:
@pierotofy would you be willing to create the project inside OpenDroneMap organization in github, or should I start it under my own?

pierotofy, 2017-08-18:
@smathermather would be the right person to answer this.

rmallermartins, 2017-08-20:
I created it in my own so I could start developing: https://github.com/rmallermartins/python-WebODM. @pierotofy I already have some questions about the API, should I ask them here?

pierotofy, 2017-08-20:
Awesome. Yes you can ask here!

rmallermartins, 2017-08-20:
Nice, will start with some then :)

**1. Are there Endpoints to retrieve, for example, one Project like "GET /api/projects/{id}"?** _I don't see any in docs, just listing with filters_
**2. Do Create, Update, Delete, etc... have json content on responses?** _Like Create returning the created Project fields, not well documented too_
**3. What the API returns when my auth Token has expired?**
**4. Is there a pattern for returning errors?** _Like both "required" and "Unable to login" returns with status_code 400. If it is the only code that return errors with, and if theses errors have some kind of pattern, or it depends on each API point?_

Ps: sorry if my English is rusty

pierotofy, 2017-08-23:
@rmallermarmartinis We've forked your repo to https://github.com/OpenDroneMap/python-WebODM. If you want to make the OpenDroneMap repo the official repo, simply destroy (via the settings panel) your repository. You should have received commit permissions to do changes directly to the OpenDroneMap repository.

rmallermartins, 2017-08-23:
@pierotofy Nice, already did that! Yes, I received permissions from @dakotabenjamin

rmallermartins, 2017-08-24:
Guys, 2 problems.

1. Since the repo changed, Travis and Pyup stopped working and I don't have admin access to enable the process with my account.
2. I think Issues are disabled in the repo, could someone enable it? So we can start using the projects issues for this kind of thing ;)

pierotofy, 2017-08-24:
1. I've enabled a travis build for that repo. I'm not too familiar with pyup, so I haven't done anything there yet. @dakotabenjamin would it be possible to grant @rmallermartins the permissions he needs to setup pyup?
2. I've enabled issues, they should be OK now.

Thanks!

pierotofy, 2020-04-02:
Closing due to lack of interest / PyODM seems the better way to go forward as far as Python support is concerned.

closed 2020-04-02 by pierotofy