WebODM/WebODM #104: job with 138 images

jfcastiglia, 2017-02-15:
I am a very new user I am running on Mac 

I get this inside webODM,

Killed
Traceback (most recent call last):
  File "/code/run.py", line 55, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/pmvs.py", line 73, in process
    (context.pmvs2_path, tree.pmvs_rec_path))
  File "/code/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 137

And this inside terminal,

webapp        | WARNING Run time of job "background.<locals>.wrapper (trigger: interval[0:00:05], next run at: 2017-02-15 21:29:26 UTC)" was missed by 0:00:01.765431
webapp        | WARNING Run time of job "background.<locals>.wrapper (trigger: interval[0:00:30], next run at: 2017-02-15 21:29:51 UTC)" was missed by 0:00:01.720077
webapp        | WARNING Run time of job "cleanup_projects (trigger: interval[0:00:15], next run at: 2017-02-15 21:29:36 UTC)" was missed by 0:00:01.767817
webapp        | INFO Acquiring lock: Task [Task of 2017-02-15T20:49:27.479Z] (1)
webapp        | INFO Processing status: 30 for Task [Task of 2017-02-15T20:49:27.479Z] (1)

pierotofy, 2017-02-16:
Hi @jfcastiglia thanks for reporting the problem. I'm going to guess that you haven't allocated enough RAM to your Docker container. Please see these instructions to increase the amount of RAM allocated: http://stackoverflow.com/a/39720010

jfcastiglia, 2017-02-16:
I gave it 2 cores and 8Gb ram this time it stoped after 25 min in will retry and post what errors I get

pierotofy, 2017-02-20:
I'll close this for now unless there are other questions.

closed 2017-02-20 by pierotofy