WebODM/WebODM #105: relation "auth_user" does not exist

mbandy, 2017-02-17:
Hi! Following the install directions all seemed to be well until I tried to log in as admin. Then this happened. Any suggestions? Thanks in advance!

> ProgrammingError at /login/
> relation "auth_user" does not exist
> LINE 1: ...user"."is_active", "auth_user"."date_joined" FROM "auth_user...
>                                                              ^
> Request Method:	POST
> Request URL:	http://10.100.103.55:8000/login/
> Django Version:	1.10.5
> Exception Type:	ProgrammingError
> Exception Value:	
> relation "auth_user" does not exist
> LINE 1: ...user"."is_active", "auth_user"."date_joined" FROM "auth_user...
>                                                              ^
> Exception Location:	/home/inti/WebODM/v_odm/local/lib/python3.4/site-packages/django/db/backends/utils.py in execute, line 64
> Python Executable:	/home/inti/WebODM/v_odm/bin/python
> Python Version:	3.4.3
> Python Path:	
> ['/home/inti/WebODM',
>  '/home/inti/WebODM',
>  '/home/inti/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages',
>  '/home/inti/OpenDroneMap/SuperBuild/src/opensfm',
>  '/home/inti/WebODM/v_odm/lib/python3.4',
>  '/home/inti/WebODM/v_odm/lib/python3.4/plat-x86_64-linux-gnu',
>  '/home/inti/WebODM/v_odm/lib/python3.4/lib-dynload',
>  '/usr/lib/python3.4',
>  '/usr/lib/python3.4/plat-x86_64-linux-gnu',
>  '/home/inti/WebODM/v_odm/local/lib/python3.4/site-packages',
>  '/home/inti/WebODM/v_odm/lib/python3.4/site-packages']
> Server time:	Fri, 17 Feb 2017 00:46:16 -0700

pierotofy, 2017-02-17:
Hi @mbandy thanks for reporting the problem.

Have you run `./start.sh`?

pierotofy, 2017-02-20:
Will close this for now unless there are other questions.

closed 2017-02-20 by pierotofy

mbandy, 2017-02-21:
Sorry to be so slow to respond, @pierotofy . To follow up on this, it turns out I had PostGIS installed under an old version of Postgresql. Once I installed it under the current version (9.6) the problem went away. 

Thanks!