WebODM/WebODM #106: Active Directory Authentication

smathermather, 2017-02-24:
@dnewcomb asked today about what is needed to allow / add active directory authentication into WebODM.

pierotofy, 2017-02-25:
I would probably integrate this one since it's Python 3 compatible: https://github.com/Lucterios2/django_auth_ldap3_ad

smathermather, 2017-02-25:
Awesome. Thanks!

smathermather, 2017-02-25:
@newcomb-d, looks like you can easily set up LDAP/AD authentication with WebODM. Cheers!

newcomb-d, 2017-02-27:
Outstanding!
Thanks!

Doug

-- 
Doug Newcomb
USFWS
551F Pylon Dr
Raleigh, NC
919-856-4520 ext. 14 doug_newcomb@fws.gov
---------------------------------------------------------------------------------------------------------
The opinions I express are my own and are not representative of the
official policy of the U.S.Fish and Wildlife Service or Dept. of the
Interior.   Life is too short for undocumented, proprietary data formats.
*NOTE: This email correspondence and any attachments to and from this
sender is subject to the Freedom of Information Act (FOIA) and may be
disclosed to third parties.*​

pvgmenegasso, 2021-04-01:
Hello to you all, has any further progress been made on this ?
I'm trying to setup LDAP authentication on WebODM but I'm kind of lost on where to start.
Thanks a lot :)

pierotofy, 2021-04-02:
No progress has been made (but we'd still welcome contributions to implement it)!

pvgmenegasso, 2021-04-02:
> No progress has been made (but we'd still welcome contributions to implement it)!

It would be a pleasure to contribute to the project !
Would you have some tips on where to start ? I'm still on the process of understanding the project's architecture. From what I've gathered it seems like I would have to temper on the Django files related to WebODM right ? Any tips would be appreciated :)

pierotofy, 2021-04-02:
Yep, Django docs would be a good place to start.

pvgmenegasso, 2021-05-21:
I still had no luck with this. Has anyone made any progress on this subject ? @newcomb-d  @smathermather  ?

smathermather, 2021-05-22:
No progress that I know of.

pierotofy, 2023-01-28:
I'm going to close this, but a PR is still welcome :pray:

closed 2023-01-28 by pierotofy