WebODM/WebODM #107: Processing node not recognized

mbandy, 2017-02-25:
Hi! Great project. I am very excited to use it. I am hoping someone can help me out.

I have a working native install of OpenDroneMap, WebODM and node-OpenDroneMap on Ubuntu 14.04. ODM works and has been tested on multiple datasets. Node-ODM also works. I can connect to it using a web browser, upload photos, and download assets. WebODM also seems to work, since I can log in, create/modify users, and so on.

Here's the probem. I can add the node-ODM instance as a processing node via WebODM and it adds sucessfully. However, if I make a new task and upload images I get the following error in a popup. I have added the proper permissions to the user/group and am at a loss as to how to proceed. Help would be much appreciated. Thanks!

> 

```
> There are no usable processing nodes. We tried to reach:
>        10.100.103.55:3000 (queue: 0)
> Make sure that at least one processing node is reachable and that you have granted the current user sufficient permissions to view the processing node (by going to Administration -- Processing Nodes -- Select Node -- Object Permissions -- Add User/Group and check CAN VIEW PROCESSING NODE). If you are bringing a node back online, it will take about 30 seconds for WebODM to recognize it.
```

pierotofy, 2017-02-25:
Hi @mbandy thanks for reporting the issue!

I'm guessing that the problem has to do with your network configuration; 10.100.103.55 is an address that must be reachable from the machine that is hosting WebODM.

| WebODM  |    ---------> |  10.100.103.55 |

The fact that you can access it independently:

| Your machine  |    ---------> |  10.100.103.55 |

Suggests that you might have to configure your firewall and/or network routes on the WebODM host.

If WebODM and node-OpenDroneMap are running on the same machine, simply use "localhost" for the IP address instead of 10.100.103.55.

mbandy, 2017-02-25:
HI, @pierotofy  thanks for your quick response!

I apologize, I should have included a bit more information in my question. I am pretty sure there is no network/routing issue. WebODM and node-ODM are running on the same machine. I have added the  processing node as localhost and also as 127.0.0.1 and WebODM exhibits the same behavior (below). And I can access 10.100.103.55:3000 from a shell environment (using telnet or a browser like links). So the machine can definitely connect to itself using any three of those network addresses. I am stuck!

```
There are no usable processing nodes. We tried to reach:
10.100.103.55:3000 (queue: 0)
localhost:3000 (queue: 0)
127.0.0.1:3000 (queue: 0)
Make sure that at least one processing node is reachable and that you have granted the current user sufficient permissions to view the processing node (by going to Administration -- Processing Nodes -- Select Node -- Object Permissions -- Add User/Group and check CAN VIEW PROCESSING NODE). If you are bringing a node back online, it will take about 30 seconds for WebODM to recognize it.
```

pierotofy, 2017-02-25:
Mm, very odd indeed! Let's try a few things to see if we can narrow it down:

1. If you open the processing node details (by visiting the entries in the Processing Nodes menu, see screenshot):

[image]

What does the `Last Refreshed` field report?

2. If you add `nodeodm.masseranolabs.com:80` (which is a public test node known to work), does the system recognize it (does the `Last Refreshed` field populate along with the `API Version`, `Queue Count`, etc.)?

3. What's the output of (from WebODM's root directory)?

```bash
for H in "localhost" "10.100.103.55"
do
echo "from nodeodm.models import ProcessingNode;p=ProcessingNode(hostname='$H',port=3000);print(p.update_node_info());print(vars(p));" | python manage.py shell
done
```

and

```
echo "import requests;print(requests.get('http://10.100.103.55:3000/info',timeout=5).json())" | python manage.py shell
```

?

mbandy, 2017-02-25:
Thanks!

1) `Last Refreshed` for all three processing nodes (127.0.0.1, localhost, 10.100.103.55) shows "0 minutes ago (Feb. 25, 2017, 9:40 a.m.)"

2) nodedm.massseranolabs.com:80 adds successfully and `Last Refreshed`, `API Version`, and `Queue Count` all populate as expected.

3) Output below. The first....
```
Python 3.4.3 (default, Nov 17 2016, 01:08:31)
[GCC 4.8.4] on linux
Type "help", "copyright", "credits" or "license" for more information.
(InteractiveConsole)
>>> True
{'available_options': [{'type': 'int', 'help': 'Oct-tree depth at which the Laplacian equation is solved in the surface reconstruction step. Increasing this value increases computation times slightly but helps reduce memory usage. Default: 9', 'value': '9', 'name': 'mesh-solver-divide', 'domain': 'positive integer'}, {'type': 'float', 'help': 'Ignore matched keypoints if the two images share less than <float> percent of keypoints.
Default: 2', 'value': '2', 'name': 'matcher-threshold', 'domain': 'percent'}, {'type':
'int', 'help': 'Oct-tree depth used in the mesh reconstruction, increase to get more vertices, recommended values are 8-12. Default: 9', 'value': '9', 'name': 'mesh-octree-depth', 'domain': 'positive integer'}, {'type': 'int', 'help': 'Minimum number of features to extract per image. More features leads to better results but slower execution. Default: 4000', 'value': '4000', 'name': 'min-num-features', 'domain': 'integer'}, {'type': 'float', 'help': 'Override the focal length information for the images', 'value': '0', 'name': 'force-focal', 'domain': 'positive float'}, {'type': 'int', 'help': 'resizes
images by the largest side', 'value': '2400', 'name': 'resize-to', 'domain': 'integer'}, {'type': 'bool', 'help': 'Keep faces in the mesh that are not seen in any camera. Default:  false', 'value': 'false', 'name': 'texturing-keep-unseen-faces', 'domain': ''},
{'type': 'int', 'help': 'The level in the image pyramid that is used for the computation. see http://www.di.ens.fr/pmvs/documentation.html for more pmvs documentation. Default: 1', 'value': '1', 'name': 'pmvs-level', 'domain': 'positive integer'}, {'type': 'float', 'help': 'A patch reconstruction is accepted as a success and kept if its associated photometric consistency measure is above this threshold. Default: 0.7', 'value': '0.7', 'name': 'pmvs-threshold', 'domain': 'float: -1.0 <= x <= 1.0'}, {'type': 'string', 'help': 'Type of photometric outlier removal method: [none, gauss_damping, gauss_clamping]. Default: gauss_clamping', 'value': 'gauss_clamping', 'name': 'texturing-outlier-removal-type', 'domain': 'string'}, {'type': 'bool', 'help': 'Skip filling of holes in the mesh. Default:  false', 'value': 'false', 'name': 'texturing-skip-hole-filling', 'domain': ''}, {'type': 'float', 'help': 'Orthophoto ground resolution in pixels/meterDefault: 20', 'value': '20', 'name': 'orthophoto-resolution', 'domain': 'float > 0.0'}, {'type': 'float', 'help': 'Override the ccd width information for the images', 'value': '0', 'name': 'force-ccd', 'domain': 'positive float'}, {'type': 'int', 'help': 'The maximum vertex count of the output mesh Default: 100000', 'value': '100000', 'name': 'mesh-size', 'domain': 'positive integer'}, {'type': 'int', 'help': 'Distance threshold in meters to find pre-matching images based on GPS exif data. Set to 0 to skip pre-matching. Default: 0', 'value': '0', 'name': 'matcher-distance', 'domain': 'integer'}, {'type': 'bool', 'help': 'Print additional messages to the console\nDefault: false', 'value': 'false', 'name': 'verbose', 'domain': ''}, {'type': 'int', 'help': 'Cell size controls the density of reconstructionsDefault: 2', 'value': '2', 'name': 'pmvs-csize', 'domain': 'positive integer'}, {'type': 'bool', 'help': 'Use this tag if you have a gcp_list.txt but want to use the exif geotags instead', 'value': 'false', 'name': 'use-exif', 'domain': ''}, {'type': 'float', 'help': 'Number of points per octree node, recommended and default value: 1', 'value': '1', 'name': 'mesh-samples', 'domain': 'float >= 1.0'}, {'type': 'bool', 'help': 'Skip local seam blending. Default:  false', 'value': 'false', 'name': 'texturing-skip-local-seam-leveling', 'domain': ''}, {'type': 'int', 'help': 'pmvs samples wsize x wsize pixel colors from each image to compute photometric consistency score. For example, when wsize=7, 7x7=49 pixel colors are sampled in each image. Increasing the value leads to more stable reconstructions, but the program becomes slower. Default: 7', 'value': '7', 'name': 'pmvs-wsize', 'domain': 'positive integer'}, {'type': 'string', 'help': 'Data term: [area, gmi]. Default: gmi', 'value': 'gmi', 'name': 'texturing-data-term', 'domain': 'string'}, {'type': 'string', 'help': 'Path to input images',
'value': '', 'name': 'images', 'domain': 'string'}, {'type': 'string', 'help': 'Path to the video file to process', 'value': '', 'name': 'video', 'domain': 'string'}, {'type': 'int', 'help': 'Each 3D point must be visible in at least minImageNum images for being reconstructed. 3 is suggested in general. Default: 3', 'value': '3', 'name': 'pmvs-min-images', 'domain': 'positive integer'}, {'type': 'int', 'help': 'The maximum number of processes to use in dense reconstruction. Default: 4', 'value': '4', 'name': 'opensfm-processes', 'domain': 'positive integer'}, {'type': 'bool', 'help': 'Skip global seam
[... cut ...] the Neighbors parameter. Default: 8', 'value': '8', 'name': 'matcher-neighbors', 'domain': 'integer', 'type': 'int'}], 'api_version': '1.0.1', 'queue_count': 0, 'last_refreshed': datetime.datetime(2017, 2, 25,
16, 46, 6, 806083, tzinfo=<UTC>), 'id': 20}
>>>
```
 ... and the second.

```
Python 3.4.3 (default, Nov 17 2016, 01:08:31)
[GCC 4.8.4] on linux
Type "help", "copyright", "credits" or "license" for more information.
(InteractiveConsole)
>>> {'taskQueueCount': 0, 'version': '1.0.1'}
>>>
```

mbandy, 2017-02-25:
Interestingly, though, `nodeodm.masseranolabs.com:80` exhibits the same behavior.

```
There are no usable processing nodes. We tried to reach:
localhost:3000 (queue: 0)
10.100.103.55:3000 (queue: 0)
nodeodm.masseranolabs.com:80 (queue: 1)
Make sure that at least one processing node is reachable and that you have granted the current user sufficient permissions to view the processing node (by going to Administration -- Processing Nodes -- Select Node -- Object Permissions -- Add User/Group and check CAN VIEW PROCESSING NODE). If you are bringing a node back online, it will take about 30 seconds for WebODM to recognize it.
```

mbandy, 2017-02-25:
Could this perhaps be a python 2/3 version issue? I am running WebODM under a virtualenv, while the main system python version is 2.7. Just a thought.

pierotofy, 2017-02-25:
The good news is that it's not a networking issue :)

No virtualenv with Python 3 alongside 2 should work fine, that's what I currently use also.

Could you paste the output of http://10.100.103.55:8000/api/processingnodes/?has_available_options=True (change 8000 with the port you are using for WebODM).

Also, what browser are you using? Perhaps this is a Javascript problem.

pierotofy, 2017-02-25:
It could also be a time problem. Is the time on both your server and computer synchronized to a time server? A difference of a few minutes could throw off the following logic:

```javascript
let now = new Date();
let last_refreshed = new Date(node.last_refreshed);
let enabled = (now - last_refreshed) < 1000 * 60 * 5; // 5 minutes
```

In which case this is a bug and I should start computing whether a node is enabled on the server exclusively instead of relying on the client's clock.

mbandy, 2017-02-25:
I am using Chrome. I just tried with Firefox, though, and it's the same behavior.

I think you may have hit on something with the time. Then server and my laptop are about 10 minutes apart. Let me try to fix that and I'll let you know the results.

Output below:

```
HTTP 200 OK
Allow: GET, POST, OPTIONS
Content-Type: application/json
Vary: Accept

[
    {
        "id": 20,
        "hostname": "10.100.103.55",
        "port": 3000,
        "api_version": "1.0.1",
        "last_refreshed": "2017-02-25T17:16:09.696361Z",
        "queue_count": 0,
        "available_options": [
            {
                "value": "9",
                "domain": "positive integer",
                "name": "mesh-solver-divide",
                "help": "Oct-tree depth at which the Laplacian equation is solved in the surface reconstruction step. Increasing this value increases computation times slightly but helps reduce memory usage. Default: 9",
                "type": "int"
            },
            {
                "value": "2",
                "domain": "percent",
                "name": "matcher-threshold",
                "help": "Ignore matched keypoints if the two images share less than <float> percent of keypoints. Default: 2",
                "type": "float"
            },
            {
                "value": "9",
                "domain": "positive integer",
                "name": "mesh-octree-depth",
                "help": "Oct-tree depth used in the mesh reconstruction, increase to get more vertices, recommended values are 8-12. Default: 9",
                "type": "int"
            },
            {
                "value": "4000",
                "domain": "integer",
                "name": "min-num-features",
                "help": "Minimum number of features to extract per image. More features leads to better results but slower execution. Default: 4000",
                "type": "int"
            },
            {
                "value": "0",
                "domain": "positive float",
                "name": "force-focal",
                "help": "Override the focal length information for the images",
                "type": "float"
            },
            {
                "value": "2400",
                "domain": "integer",
                "name": "resize-to",
                "help": "resizes images by the largest side",
                "type": "int"
            },
            {
                "value": "false",
                "domain": "",
                "name": "texturing-keep-unseen-faces",
                "help": "Keep faces in the mesh that are not seen in any camera. Default:  false",
                "type": "bool"
            },
[... 629 lines cut ...]
            {
                "value": "0.6",
                "domain": "float",
                "name": "matcher-ratio",
                "help": "Ratio of the distance to the next best matched keypoint. Default: 0.6",
                "type": "float"
            },
            {
                "value": "false",
                "domain": "",
                "name": "use-pmvs",
                "help": "Use OpenSfM to compute the point cloud instead of PMVS",
                "type": "bool"
            },
            {
                "value": "8",
                "domain": "integer",
                "name": "matcher-neighbors",
                "help": "Number of nearest images to pre-match based on GPS exif data. Set to 0 to skip pre-matching. Neighbors works together with Distance parameter, set both to 0 to not use pre-matching. OpenSFM uses both parameters at the same time, Bundler uses only one which has value, prefering the Neighbors parameter. Default: 8",
                "type": "int"
            }
        ]
    }
]
```

mbandy, 2017-02-25:
@pierotofy,  you're a genius! It was the time. I fixed the server time and everything now works as expected. Thank you! I really appreciate the help!

pierotofy, 2017-02-25:
Glad it worked!

I've marked this as a bug, since I can probably compute the time difference on the server side thus without the need for a synchronized clock. I'll leave the issue open until I have a fix for it.

Thanks for reporting the problem!

pierotofy, 2017-03-06:
This is now fixed in the `bugfixes` branch: https://github.com/pierotofy/WebODM/commit/6276ebbe18f9f635a6d1d02d6b35e21a84f21cb0

Will land into master sometimes later this or next week.

commit 6276ebbe (2017-03-08): Added online field on processing nodes, which fixes #107
closed 2017-03-06 by pierotofy