WebODM/WebODM #109: Process exited with code 1

mtaivalmaa, 2017-02-28:
I have installed webodm using docker on my Macbook air i7 8gb ..

I have successfully run a project with 14 4K photos from my DJI Mavic no issues.

When I load a larger project 90 - 120 4K pics, I get the following after about 15mins. of processing.

Any help would be great.

DoG running...4231 dog done
4148 dog done
done
Killed
Traceback (most recent call last):
  File "/code/run.py", line 55, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/pmvs.py", line 73, in process
    (context.pmvs2_path, tree.pmvs_rec_path))
  File "/code/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 137

pierotofy, 2017-02-28:
Hi @mtaivalmaa, per https://github.com/OpenDroneMap/WebODM#common-troubleshooting if you see a `Killed` message it means you are running out of RAM.

See this link on how to increase it: http://stackoverflow.com/a/39720010

mtaivalmaa, 2017-02-28:
I was wondering about that .. thank you!

closed 2017-02-28 by pierotofy

mtaivalmaa, 2017-02-28:
Per pierotofy.. 

Setting Docker from 2gig ram to 4 gig ram fixed my issue.

See this link on how to increase it: http://stackoverflow.com/a/39720010

Kind Regards,

Mike