WebODM/WebODM #110: When tasks are moved in the database the filesystem is not updated

mbandy, 2017-03-01:
I have a working install for the use of my group and it is great. Thanks so much for this wonderful software!

I think this can be classified as a bug. When a task is reassigned to a different project in the administrative interface, the assets are not moved in the filesystem (app/media/project). So the assets (and source photos) are no longer available via the web interface. It's simple enough to move them by hand in the shell (after which everything works as expected) but WebODM should probably take care of this behind the scenes. 

Thanks again!

pierotofy, 2017-03-01:
Yes this is definitely a bug, along with #108. I will put priority on both of these in the next week or two.

Thank you for reporting the problem!

pierotofy, 2017-03-03:
This is fixed here https://github.com/pierotofy/WebODM/commit/0fa38e310bbb2f3bdeaef53e188f8f1441e77fb3, changes which I hope to land into master next week.

closed 2017-03-03 by pierotofy

pierotofy, 2017-03-03:
Note that the branch still requires more testing, I wouldn't merge it just yet.