WebODM/WebODM #112: Exception: Child returned 134

mtaivalmaa, 2017-03-01:
I just installed WebODM on ubuntu using docker.. was running a small test only 6 files.  got the following.. 

-Mike

Generating texture patches:
	Running... terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)
Traceback (most recent call last):
  File "/code/run.py", line 55, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/mvstex.py", line 106, in process
    '{keepUnseenFaces}'.format(**kwargs))
  File "/code/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 134

pierotofy, 2017-03-01:
It's likely that 6 files are not enough to identify enough features to generate an orthophoto. Would you be able to share your dataset? Perhaps there are a few options that could be tweaked to make it work.

mtaivalmaa, 2017-03-01:
Ok, I was wondering.. I am going to try a full 90 images.. Its been working on my Macbook air but I wanted more processing power.. 
Thanks

closed 2017-03-01 by mtaivalmaa