WebODM/WebODM #113: Contour Maps with point annotations

hblanken, 2017-03-04:
It would be great to have a 2D/3D contour map (with option for different step widths) and step height labels attached to the contour lines and then the ability to select points on the 2d/3d map to mark as x,y,z with txt label. Similar to a professional RE survey. The heights could be relative to above-sea-level (tbd if possible?). E.g. Roof Ridge = RR (23ft5in) or RR (7m15cm), TGUT (5m10cm). 2D map would be great, 3D would be extra cool.

hblanken, 2017-03-15:
More flavour to this - it is related to GRASS script  - has anyone tested whether this already works on ODM now? https://github.com/OpenDroneMap/OpenDroneMap/tree/master/contrib/grass

pierotofy, 2017-03-15:
I haven't tried it yet. Perhaps somebody could report on how well it works!

smathermather, 2017-03-15:
I haven't gotten a chance to test it yet, but will say AFAIK, this doesn't do point cloud filtering, so the contours would be DSM contours, not DTM contours. I'll see if I can test soon.

yawstick, 2018-09-10:
Still a newbie ere and hardly know enough to ask intelligent questions. I did manage to get WebODM running on a cloud server and trying to get up to speed on all the terminology. My interest is primarily subdivision construction related. It would be nice to have contour maps to provide to the folks who estimate grading and civil engineering costs for a given site. Thanks for great work... I come from the drone community and have been involved with drones since the very early days when they were using Arduino based controllers.... come a long way since then.

pierotofy, 2019-01-07:
+1 for this (recording user feedback).

ChipwizBen, 2019-02-26:
Another +1 from me. Somebody just requested this.

pierotofy, 2019-03-02:
We've been having quite a bit of interest to implement this. Perhaps with everyone's financial help we could give it a push.

I'm launching a mini-campaign to see if we can raise funds to implement this as a community. If you can commit some funds to this feature, please fill-in this form: https://goo.gl/forms/U10TiSOazkLmqKmA3

pierotofy, 2019-04-02:
Contours preview and export are now implemented in the 2D map as part of #639. I will close this and open a new one for tracking point annotations.

closed 2019-04-02 by pierotofy