WebODM/WebODM #115: webodm update deletes all projects in DB

hblanken, 2017-03-08:
Objective is to keep DB data persistent between updates. Currently the update command will nuke the db. I think this is an issue. Docker containers are not really designed to keep persistent data (although some people do it anyways). The best thing at the moment to persist data between updates would be do run WebODM / PostGIS on a local machine, and use docker to just run node-opendronemap. But there might be a way around it using docker volumes...or another solution to be investigated.

pierotofy, 2017-03-08:
After doing some initial research, it should be possible to maintain volume persistence by defining a named volume for the DB instance. 

@hblanken since you experienced this problem, are there any files left from your previous projects in ~/webodm/app/media/project ?

hblanken, 2017-03-08:
I might be looking in the wrong place - but my media folder is empty when I go through terminal.

hblanken, 2017-03-08:
I might be looking in the wrong way. I am running on Ec2 on docker - My app/media/project folder is completely empty - actually the folder does not exist.

I have found the old files here:
/var/lib/docker/volumes/517c3b5801359e483c12f516570af10c5978004c440ca511ca3af3d7915fc75a/_data/project/2/

and my new projects are here: 
/var/lib/docker/aufs/mnt/ff7a2fa1dd1657f7643e176191bdf4f4e5b747d29dbb7fddfe82ec9f5f79e059/var/www/data

pierotofy, 2017-03-08:
This is now solved in https://github.com/pierotofy/WebODM/commit/9b3f0843d0363cda92284a2e491883f9d40d70c4

I will fast-track the merge of this branch since it probably caused other users to see their data disappear. Note that the data is still in a docker volume and hasn't been lost, it just hasn't been mounted back after update.

pierotofy, 2017-03-08:
@hblanken you if update webodm the database will reset once more, but subsequent updates should maintain data persistently.

Please let me know if the problem happens again! Thanks for reporting the problem.

closed 2017-03-08 by pierotofy

smathermather, 2017-03-09:
Ahh, this is nice... .