WebODM/WebODM #117: Process exited with code null

tripplowe, 2017-03-09:
I have a fresh Ubuntu 16.04 server install on which I have a working installation of Docker and Docker-compose, and OpenDroneMap - I did not use the Docker/OpenDroneMap installation method.  

I have WebODM installed but each job exits (shortly after it starts) with a 'Process exited with code null' error.  This is the ODM code generated from my last attempt:

`python /code/run.py --verbose --use-exif --images /home/biomata/odm_data_toledo/ --opensfm-processes 24 --use-pmvs --project-path /var/www/data/590007ca-c11f-4a46-9320-6a1d40a46e9c --pmvs-num-cores 32
`
Is WebODM trying to save the dataset I upload to /var/www/data/...?  I just checked and there is no /var/www folder - could this be a problem, if so, how should I set the permissions?  I changed the --project-path from /var/www/data/... to the odm_data-toledo directory on my desktop and successfully ran it in ODM.  

Is there an advanced setting where I can specify where the uploaded files are to be saved?

Thanks.
-Tripp

pierotofy, 2017-03-09:
Hi @tripplowe, are you using WebODM via docker? What about node-odm?

Also, is there any console output from the task besides the "Process exited with code null" error?

tripplowe, 2017-03-09:
Yes, I am running WebODM in docker.  I followed the Getting Started instructions on github.com/OpenDroneMap/WebODM so node-odm is started in docker too, right?

There is no output returned to the WebODM console.  The terminal output reports

- node-odm-1  info:  about to run: python / code/run.py...
- webapp INFO Acquiring lock: task [...]
- webapp INFO Processing status: 30 for Task...

pierotofy, 2017-03-09:
Yes node-odm-1 is also started. Note that node-odm-1 is not the same OpenDroneMap installation you are natively running on your Ubuntu server, it's a dockerized and separate install of OpenDroneMap (if this sounds confusing, bear with me).

You have OpenDroneMap installed on your machine, natively.

You have WebODM installed via Docker.

WebODM by default creates a separate OpenDroneMap installation (node-odm-1) to help users get started. This is the one you are currently using within WebODM.

If you want to use your native OpenDroneMap installation with WebODM, you first need to add node-OpenDroneMap (https://github.com/OpenDroneMap/node-OpenDroneMap#run-it-natively), then add a new processing node within WebODM (and remove node-odm-1).

But the reason your task is failing, is because you are specifying the "--images" option. You don't need to do that. Just upload the images to the WebODM dashboard, and WebODM will take care of the rest.

tripplowe, 2017-03-09:
I get the exit code null with and without the --image option.  I will add node-OpenDroneMap and will try again.  Thanks.

tripplowe, 2017-03-09:
I get several warnings when installing npm
```
npm WARN deprecated node-uuid@1.4.7: use uuid module instead
npm WARN prefer global coffee-script@1.10.0 should be installed with -g
npm WARN prefer global jshint@2.9.4 should be installed with -g
npm WARN deprecated node-uuid@1.4.7: use uuid module instead
npm WARN prefer global coffee-script@1.10.0 should be installed with -g
npm WARN prefer global jshint@2.9.4 should be installed with -g

```
Is this going to be an issue?

pierotofy, 2017-03-09:
Nop, you should be all right :)

tripplowe, 2017-03-09:
Thank you for your help thus far.  I have node-OpenDroneMap up and running a job now at localhost:3001.    I tried to add 127.0.0.1:3001 as a processing node in WebODM, but it seems to be offline.  Is there any way to add this to the WebODM interface?

pierotofy, 2017-03-09:
You need to use the IP address of a network interface on your host. Most of the confusion here is due to the fact that WebODM is running within a docker container. If you specify "localhost", or 127.0.0.1, WebODM thinks that node-odm is running within the docker container.

On your Ubuntu host, do a `ifconfig -a` and find the IP address assigned to your network card, then use that IP instead.

I'd highly recommend reading more about docker if you are going to use WebODM in the long run :)

tripplowe, 2017-03-09:
Holy cow, that worked (I don't mean to sound surprised).  Thank you again for your assistance.

pierotofy, 2017-03-09:
Yay!

closed 2017-03-09 by pierotofy