WebODM/WebODM #121: Normalize Assets Paths

pierotofy, 2017-03-16:
Right now assets are all over the place. node-opendronemap should create a strict standard of where assets should be placed before being downloaded.

For example:

`/pointcloud.las`
`/orthophoto.tif`
`/texturedmodel/model.obj
`/texturedmodel/model.mtl
`/dem.tif`
`/contours.shp`

This will make it easier to future proof paths and perhaps allow other photogrammetry engines to plug-in into WebODM.

smathermather, 2017-03-16:
Agreed. I know @dakotabenjamin has given this some thought.

pierotofy, 2020-04-02:
Albeit this was a good idea, it would add significant complexity. I'm going to take this off the TODO list as existing filename conventions seem to be holding pretty well. :clinking_glasses:

closed 2020-04-02 by pierotofy