WebODM/WebODM #122: Process exited with code 1

LeneFischer, 2017-03-19:
Hi
My first question here :-) 
Is this the correct way of asking a question?

Second question:
I start WebODM on Linux and everything run smoothly - Login, Upload images (11 images) and a lot of processing. It all ends with 'Process exited with code 1'

I have tried to look for exception codes - with no luck  - so I ask here.

Any suggestion on what I might have done wrong.
Regards
Lene Fischer

The dialog looks like this long text:

[INFO]    Initializing OpenDroneMap app - Sun Mar 19 07:26:10  2017
[INFO]    Running ODM Load Dataset Cell
[DEBUG]   resize dir: /var/www/data/9306309d-ad93-4cf4-9156-f8f8f98249f0/images_resize
[DEBUG]   Loading dataset from: /var/www/data/9306309d-ad93-4cf4-9156-f8f8f98249f0/images_resize
[DEBUG]   /var/www/data/9306309d-ad93-4cf4-9156-f8f8f98249f0/images_resize
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0017.JPG | camera: dji fc6310 | dimensions: 2400 x 1600 | focal: 8.8 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0026.JPG | camera: dji fc6310 | dimensions: 2400 x 1600 | focal: 8.8 | ccd: None

[DEBUG]   Loaded DJI_0023.JPG | camera: dji fc6310 | dimensions: 2400 x 1600 | focal: 8.8 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0022.JPG | camera: dji fc6310 | dimensions: 2400 x 1600 | focal: 8.8 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0020.JPG | camera: dji fc6310 | dimensions: 2400 x 1600 | focal: 8.8 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0018.JPG | camera: dji fc6310 | dimensions: 2400 x 1600 | focal: 8.8 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0021.JPG | camera: dji fc6310 | dimensions: 2400 x 1600 | focal: 8.8 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0025.JPG | camera: dji fc6310 | dimensions: 2400 x 1600 | focal: 8.8 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0027.JPG | camera: dji fc6310 | dimensions: 2400 x 1600 | focal: 8.8 | ccd: None
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0024.JPG | camera: dji fc6310 | dimensions: 2400 x 1600 | focal: 8.8 | ccd: None
[INFO]    Found 11 usable images
[INFO]    Running ODM Load Dataset Cell - Finished
[INFO]    Running ODM Resize Cell
[DEBUG]   Resizing dataset to: /var/www/data/9306309d-ad93-4cf4-9156-f8f8f98249f0/images_resize
[WARNING] Already resized DJI_0026.JPG | dimensions: 2400 x 1600
[WARNING] Already resized DJI_0017.JPG | dimensions: 2400 x 1600
[WARNING] Already resized DJI_0023.JPG | dimensions: 2400 x 1600
[WARNING] Already resized DJI_0019.JPG | dimensions: 2400 x 1600
[WARNING] Already resized DJI_0022.JPG | dimensions: 2400 x 1600
[WARNING] Already resized DJI_0018.JPG | dimensions: 2400 x 1600
[WARNING] Already resized DJI_0025.JPG | dimensions: 2400 x 1600
[WARNING] Already resized DJI_0024.JPG | dimensions: 2400 x 1600
[WARNING] Already resized DJI_0027.JPG | dimensions: 2400 x 1600
[WARNING] Already resized DJI_0020.JPG | dimensions: 2400 x 1600
[WARNING] Already resized DJI_0021.JPG | dimensions: 2400 x 1600
[INFO]    Resized 11 images
[INFO]    Running ODM Resize Cell - Finished
[INFO]    Running ODM OpenSfM Cell
[WARNING] Found a valid OpenSfM file in: /var/www/data/9306309d-ad93-4cf4-9156-f8f8f98249f0/opensfm/reconstruction.json
[WARNING] Found a valid Bundler file in: /var/www/data/9306309d-ad93-4cf4-9156-f8f8f98249f0/opensfm/reconstruction.json
[INFO]    Running ODM OpenSfM Cell - Finished
[INFO]    Running ODM Meshing Cell
[WARNING] Found a valid ODM Mesh file in: /var/www/data/9306309d-ad93-4cf4-9156-f8f8f98249f0/odm_meshing/odm_mesh.ply
[INFO]    Running ODM Meshing Cell - Finished
[INFO]    Running MVS Texturing Cell
[DEBUG]   Writing MVS Textured file in: /var/www/data/9306309d-ad93-4cf4-9156-f8f8f98249f0/odm_texturing/odm_textured_model.obj
[DEBUG]   running /code/SuperBuild/install/bin/texrecon /var/www/data/9306309d-ad93-4cf4-9156-f8f8f98249f0/opensfm/reconstruction.nvm /var/www/data/9306309d-ad93-4cf4-9156-f8f8f98249f0/odm_meshing/odm_mesh.ply /var/www/data/9306309d-ad93-4cf4-9156-f8f8f98249f0/odm_texturing/odm_textured_model -d gmi -o gauss_clamping -t none     
/code/SuperBuild/install/bin/texrecon (built on Mar 18 2017, 00:54:03)
Load and prepare mesh: 
	Could not load mesh: File format not recognized as PLY-model
Traceback (most recent call last):
  File "/code/run.py", line 55, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/mvstex.py", line 109, in process
    '{keepUnseenFaces}'.format(**kwargs))
  File "/code/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1

pierotofy, 2017-03-19:
Hi Lene, yes this is the right place to ask questions.

Since you closed the issue, does that mean that you solved the problem? Or do you still need help? In that case, would it be possible for you to share the dataset (images) you are using?

LeneFischer, 2017-03-20:
Hi Piero,
I closed the issue because I found an answer, that indicated it was RAM that was the issue.
But, haven´t solved the issue yet.

Regards

Lene Fischer
Associate Professor

University of Copenhagen
Department of Geoscience and Natural Resource Management

Forest and Landscape College
Nødebovej 77a
3480
Denmark

TEL +45 40115084
lfi@ign.ku.dk<mailto:lfi@ign.ku.dk>

[cid:image004.gif@01D2A143.DB876510]

[cid:image002.png@01D2A0D7.90033DB0]<@Lene_Fischer> [cid:image003.png@01D2A0D7.90033DB0] <ku-gis.dk/blog>

pierotofy, 2017-03-20:
It really depends on the size of your dataset and what settings you are using. I can run 500+ images datasets on my laptop which has 24 GB ram and 2 cores.

But the error you've experienced might be caused by something other than low memory. Is it possible for you share the dataset with us either publicly or privately?

DominikTom, 2017-03-27:
Hi, 
At first sorry for my english, I
[image]
 have the same problem on WebODM "Process exited with code 1" I work on iMac 256GB SSD, 16GB RAM, 2,7 GHz Intel Core i5 and Intel Iris Pro 1536 MB, Maybe you know where is the problem? I do not have idea :(

".....
2017-03-27 15:40:29,542 Cleaning depthmap for image DJI_0051.JPG
2017-03-27 15:40:30,449 Cleaning depthmap for image DJI_0039.JPG
2017-03-27 15:40:31,331 Cleaning depthmap for image DJI_0072.JPG
2017-03-27 15:40:32,048 Merging depthmaps
Killed
Traceback (most recent call last):
  File "/code/run.py", line 55, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/opensfm.py", line 90, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/code/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 137
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 137"

In agisoft photoscan pro and Pix4d everything looks good with the same photos from Dji Phantom 3 pro
Best Regards 
Dominik

pierotofy, 2017-03-27:
@DominikTom, notice the `Killed` in the console log; it means your machine ran out of memory. Make sure you have assigned your docker container enough RAM: http://stackoverflow.com/a/39720010

DominikTom, 2017-03-27:
Thank you very much! Now It works!
But...Quality is low compared to other programs :( Is it possible to change the quality of orthophotomaps?

pierotofy, 2017-03-27:
We're working to improve the output results. In the meanwhile, if you could open a new issue in the OpenDroneMap repository (https://github.com/OpenDroneMap/OpenDroneMap) with a few screenshots comparing the results you are getting (compared vs other programs) we might be able to provide additional suggestions to improve the output, or at least get a sense of what we need to focus our efforts to.

Thanks!!

smathermather, 2017-03-27:
If you have datasets to share too, those are welcome.

LeneFischer, 2017-03-28:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

DominikTom, 2017-03-28:
COMPARE PHOTOSCAN vs PIX4D vs DRONEDEPLOY vs MAPS MADE EASY VS WEBODM
https://drive.google.com/drive/folders/0B7BPgMvhrhZHSjRqM3NKb0dWQlU?usp=sharing

pierotofy, 2017-03-28:
@DominikTom most of the distortion in the WebODM orthophoto is due to the poisson reconstruction mesh. We are working on switching to use a 2.5D mesh (see https://github.com/OpenDroneMap/OpenDroneMap/issues/521) which will increase the orthophoto quality by quite a bit.

LeneFischer, 2017-03-28:
Sorry for opening and closing and opening. Clicked on the wrong button :-O

pierotofy, 2017-03-28:
No worries @LeneFischer 

I just processed the 8 images that you attached, without issues:

[image]

[image]

How are you running WebODM? Did you set it up using docker?

LeneFischer, 2017-03-28:
This looks great - yes I installed on Ubuntu and Docker as explained on the frontpage.
I will try to install on another computer. Thanks for your help.
Another question: Have you made a docker installation to add to the Live-OSGEO ?

pierotofy, 2017-03-28:
We have not, but this would be a great thing to do. @smathermather would it be too early to ask inclusion of WebODM/OpenDroneMap into Live-OSGEO? https://wiki.osgeo.org/wiki/Live_GIS_Add_Project#What_gets_into_OSGeoLive.3F

LeneFischer, 2017-03-28:
I hope you will - I´m giving a workshop at FOSS4G in Boston in August using WebODM and QGIS. So my question is relevant in this setup :-)

dakotabenjamin, 2017-03-29:
I think that would help us with our workshop as well!

smathermather, 2017-03-29:
What kind of lead time do we need for including ODM in OSGEO live? I'm hoping we can get some of the clustering / scaling + memory and processor use improvements in before we work on getting it into live.

LeneFischer, 2017-03-29:
I have just sent a question to the workshop organizer.

Regards
Lene

clarkerz, 2017-04-13:
So pleased to have finally got ODM and ODMWeb working. I feel a bit embarrassed asking this as have been searching for this all day but I am getting the warning whilst processing images as there is no CCD info in my Phantom 4 Images. What is the impact of that to the end the result and should I manually add it using ---force-ccd but I can't find that using WbeODM. Apologies in advance is this is stupid but I just cannot find it. Thanks for everything you guys are doing with ODM and WEBODM!!...Clarkerz

clarkerz, 2017-04-13:
Fantastic...thanks for that. It's been running for 1 /34 now with 67 photos and appears to be showing the same message now over the last hour... oh well..thought it was too good :)
webapp        | INFO Acquiring lock: Task [Task of 2017-04-13T13:45:56.369Z] (1)
webapp        | INFO Acquiring lock: Task [Task of 2017-04-13T13:45:56.369Z] (1)

pierotofy, 2017-04-30:
Since the original issue seems resolved, I'll close this for now.

closed 2017-04-30 by pierotofy