WebODM/WebODM #124: Elevation View trim outliers

dakotabenjamin, 2017-03-20:
When there's some noise in the elevation view, it makes it less useful:

[image]

It might be helpful to trim something like 2% of the max and min elevation points.

pierotofy, 2017-03-20:
This could probably be done using the newer version of potree.

[image]

http://potree.entwine.io/data/st-helens.html

I think the source for the new version is here, https://github.com/potree/potree/tree/develop, but I'll have to verify. With the update we would get a lot more goodies as far as the interface goes.

pierotofy, 2017-03-20:
Work in progress!

[image]

smathermather, 2017-03-21:
Is that what I think... ?

pierotofy, 2017-03-23:
Solved as part of #128

closed 2017-03-23 by pierotofy